open62541/open62541 #836: Wrong check of NodeClass

Pro, 2016-10-06:
@jpfr in Commit https://github.com/open62541/open62541/commit/dd23e5cb9206823fa2411eb18a5ae7f77b9093fb you added the check `item->targetNodeClass != node->nodeClass`

I think your change breaks the information model generation and this check should not be there?!

Assume the following steps:

The following code is called for Node 0:11624 (NamespaceFile) which is of Type 0:11595 (AddressSpaceFileType).
AddressSpaceFileType hasComponent 0:11615 (ExportNamespace) which is a method.

So when the NamespaceFile Object is created, the ExportNamespace method should be added as a reference to the new object.

https://github.com/open62541/open62541/blob/794b2c1ed3084b838cf08f7d1aa24b04bd14807f/src/server/ua_services_nodemanagement.c#L149

```
static UA_StatusCode
instantiateNode(UA_Server *server, UA_Session *session, const UA_NodeId *nodeId,
                UA_NodeClass nodeClass, const UA_NodeId *typeId,
UA_InstantiationCallback *instantiationCallback) {
	// ...
	// nodeId = 0:11624
    /* Copy members of the type and supertypes */
    for(size_t i = 0; i < hierarchySize; i++)
	// hierarchy[0] is AddressSpaceFileType
	retval |= copyChildNodesToNode(server, session, &hierarchy[i], nodeId, instantiationCallback);
	// ...
}
```

It calls
https://github.com/open62541/open62541/blob/794b2c1ed3084b838cf08f7d1aa24b04bd14807f/src/server/ua_services_nodemanagement.c#L247
```
static UA_StatusCode
copyChildNodesToNode(UA_Server* server, UA_Session* session, 
                     const UA_NodeId* sourceNodeId, const UA_NodeId* destinationNodeId, 
UA_InstantiationCallback* instantiationCallback) {
	// ...
	// sourceNodeId = 0:11595, destinationNodeId= 0:11624
	
            /* New node in child */
            if(rd->nodeClass == UA_NODECLASS_METHOD) {
                /* add a reference to the method in the objecttype */
                UA_AddReferencesItem newItem;
                UA_AddReferencesItem_init(&newItem);
                newItem.sourceNodeId = *destinationNodeId; // destinationNodeId = 0:11624 of UA_NODECLASS_OBJECT
                newItem.referenceTypeId = rd->referenceTypeId;
                newItem.isForward = true;
                newItem.targetNodeId = rd->nodeId;
                newItem.targetNodeClass = UA_NODECLASS_METHOD;
                retval = Service_AddReferences_single(server, session, &newItem);
            } else
	// ...
}

```

This calls then 
https://github.com/open62541/open62541/blob/794b2c1ed3084b838cf08f7d1aa24b04bd14807f/src/server/ua_services_nodemanagement.c#L956
```
UA_StatusCode
Service_AddReferences_single(UA_Server *server, UA_Session *session,
const UA_AddReferencesItem *item) {
	// ...
    retval = UA_Server_editNode(server, session, &item->sourceNodeId,
(UA_EditNodeCallback)addOneWayReference, item);
	// ...
}

```

Which then calls:
https://github.com/open62541/open62541/blob/794b2c1ed3084b838cf08f7d1aa24b04bd14807f/src/server/ua_services_nodemanagement.c#L933
```
static UA_StatusCode
addOneWayReference(UA_Server *server, UA_Session *session,
                   UA_Node *node, const UA_AddReferencesItem *item) {
    if(item->targetNodeClass != UA_NODECLASS_UNSPECIFIED &&
       item->targetNodeClass != node->nodeClass)
	return UA_STATUSCODE_BADNODECLASSINVALID;
	// ...
}

```

So it will fail, because item.sourceNodeId is of type UA_NODECLASS_OBJECT and item->targetNodeClass is of type UA_NODECLASS_METHOD.

Pro, 2016-10-06:
Or should the code in  `copyChildNodesToNode(...)` changed to not set the `targetNodeClass`? I.e.:
```
newItem.targetNodeClass = UA_NODECLASS_UNSPECIFIED;
```

jpfr, 2016-10-07:
Of course, checking the same nodeclass in both directions is bogus.
I changed the behaviour to be close to the standard and cover existing code.

The targetnodeclass is only checked for the first direction.
For the inverse direction, NODECLASS_UNSPECIFIED is taken. 

https://github.com/open62541/open62541/blob/3c78decad400d9ed745e8b7a72e8e89c8356b409/src/server/ua_services_nodemanagement.c#L999

Does that work for you @Pro ?

jpfr, 2016-10-07:
Btw, NODECLASS_UNSPECIFIED = 0 is the default. That's why only the method case failed and the unit tests were not triggered.

Pro, 2016-10-07:
I'll check it out and reopen the issue if it doesn't work.
Thanks!

Pro, 2016-10-07:
@jpfr There seems to be a problem with memory reallocation.

If I use `-DUA_ENABLE_GENERATE_NAMESPACE=ON` and then `make && valgrind "--error-exitcode=1" "--quiet" "--trace-children=yes" "--leak-check=full" ./server_instantiation`
I get (on my current branch which includes the two PRs from today https://github.com/Pro/open62541/tree/forte)

```
==28686== Invalid free() / delete / delete[] / realloc()
==28686==    at 0x4C2FD5F: realloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==28686==    by 0x41EB26: addOneWayReference (ua_services_nodemanagement.c:944)
==28686==    by 0x411695: UA_Server_editNode (ua_server_utils.c:263)
==28686==    by 0x41ECCC: Service_AddReferences_single (ua_services_nodemanagement.c:968)
==28686==    by 0x41F1DB: UA_Server_addReference (ua_services_nodemanagement.c:1098)
==28686==    by 0x4E1623: ua_namespaceinit_generated (ua_namespaceinit_generated.c:20411)
==28686==    by 0x40BCED: UA_Server_new (ua_server.c:898)
==28686==    by 0x401595: main (server_instantiation.c:28)
==28686==  Address 0x56a8ae0 is 0 bytes inside a block of size 240 free'd
==28686==    at 0x4C2EDEB: free (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==28686==    by 0x41FA52: deleteOneWayReference (ua_services_nodemanagement.c:1225)
==28686==    by 0x411695: UA_Server_editNode (ua_server_utils.c:263)
==28686==    by 0x41EE95: Service_AddReferences_single (ua_services_nodemanagement.c:1029)
==28686==    by 0x41F1DB: UA_Server_addReference (ua_services_nodemanagement.c:1098)
==28686==    by 0x4CFEB5: ua_namespaceinit_generated (ua_namespaceinit_generated.c:18516)
==28686==    by 0x40BCED: UA_Server_new (ua_server.c:898)
==28686==    by 0x401595: main (server_instantiation.c:28)
==28686==  Block was alloc'd at
==28686==    at 0x4C2DB8F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==28686==    by 0x4C2FDEF: realloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==28686==    by 0x41EB26: addOneWayReference (ua_services_nodemanagement.c:944)
==28686==    by 0x411695: UA_Server_editNode (ua_server_utils.c:263)
==28686==    by 0x41ECCC: Service_AddReferences_single (ua_services_nodemanagement.c:968)
==28686==    by 0x41F1DB: UA_Server_addReference (ua_services_nodemanagement.c:1098)
==28686==    by 0x4CFEB5: ua_namespaceinit_generated (ua_namespaceinit_generated.c:18516)
==28686==    by 0x40BCED: UA_Server_new (ua_server.c:898)
==28686==    by 0x401595: main (server_instantiation.c:28)
==28686== 

```

jpfr, 2016-10-07:
This particular issue fixed in 1a86fc28f064b344c3bc42953aa4eeff06f2d5f2.

However, there are loads of errors during the server creation with the generated namespace.
The first is a UA_STATUSCODE_BADTYPEDEFINITIONINVALID. Afterwards, many badparentnotfound and similar.

By the way, many of these issues and also the correct ordering are fixed in #739.
There, namespace generation for bootstrapping works just fine.
But the branch needs a rebase on master.

ichrispa, 2016-10-11:
Hi @Pro,

I simply removed the if-clause in question. Tests are still passing and travis seems happy. @jpfr: Why do we need to check node classes when creating references in the first place!? Why is it manditory at all to have at least one UNSPECIFIED nodeclass as a reference target!?

The issue referenced by @jpfr is a separate issues, see #835.

Best regards,

ichrispa

closed 2016-10-11 by ichrispa

jpfr, 2016-10-11:
The spec says..
> targetNodeClass: NodeClass of the TargetNode. The Client shall specify this since the TargetNode
might not be accessible directly by the Server.

So it's a "shall" feature specified in the spec for the AddNodes service.
**And that is why there is a member called targetNodeClass in UA_AddReferencesItem.**

>  Why is it manditory at all to have at least one UNSPECIFIED nodeclass as a reference target!?

There seems to be a misunderstanding.
If the expected targetNodeClass is UA_NODECLASS_UNSPECIFIED (zero enum, the default), then we **do not check** if the target's nodeclass matches. The check always passes.
The user deliberately needs to specify a targetNodeClass. Then we compare.

References are two-way. Of course, we might want to add a method-node to an object-node. So we set the targetNodeClass to unspecified for adding the inverse direction. So the check never fails here.

It's a required feature. And the default (UA_NODECLASS_UNSPECIFIED) is the permissive case that never hurts anyone.
Ok to re-add?