open62541/open62541 #840: how to use subscription ? the callback method donest work

xty19, 2016-10-08:
refering to  [**cleint.c**](https://github.com/open62541/open62541/blob/479803f627fa9f9387ad220f74616bfa8160f883/examples/client.c) 
(the examples in the examples folder),
i use it to monitor some variables,
i use the executive file,
change the variables via Ua-Expert, 
but the callback mehod cant work--**handler_TheAnswerChanged(UA_UInt32 monId,UA_DataValue *value,void *context)** just this method.

jpfr, 2016-10-08:
1. If I run /examples/server and /examples/client, the callback to TheAnswerChanged is working.
In order to continuously track fo changes, you need to call UA_Client_Subscriptions_manuallySendPublishRequest in a loop (with an interval) since the client is not yet asynchronous...

2. 
> and what does it mean Client tries to call a service without a session

I don't know where you got this error. The client needs to make four steps to get a session.
HEL/ACK, OpenSecureChannel, CreateSession, ActivateSession.
This is all done in the background when you call UA_Client_connect.

xty19, 2016-10-08:
@jpfr thanks for your reply and i am from china, now its time to get bed since i see your reply,

1.loop and asynchronous i will try tomorrow.
2. when i use client to visit the service ,and the service console print **info/server Client tries to call a service with out a session**

xty19, 2016-10-10:
@jpfr thankyou I made It ,iuse it in** a loop** ,and it can subscribe the result,and the callback work.

closed 2016-10-10 by jpfr