open62541/open62541 #843: 'malloc.h' not available in CLANG under Windows

escherstair, 2016-10-11:
This issue has been partially discussed in #834 but the necessary fix has not been applied to the sources.

After a deeper investigation, seems that the malloc.h header file is not one of ANSI C standard header files, but it is associated with the Microsoft C Run-Time Libraries (CRT). Probably under Windows it is available if you compile using MS Visual Studio, but it's not if you compile using CLANG.
I think that the necessary fix for thee original issue of #834 ('malloc.h' not found) is to skip the inclusion of malloc.h if CLANG is used.

I suggest to add an `#ifndef` section
```
#ifndef __clang__
# include <malloc.h>
#endif
```
around the inclusions of malloc.h

ichrispa, 2016-10-14:
Could you try to substitute it for #include <stdlib.h>? No idea if this works on windows though...

escherstair, 2016-10-17:
Probably I didn't explain very well the situation: this is not a request for help.
It's a request to modify the code of open62541 library so that it compiles with CLANG under Windows OS.

- the single pair of header and source always includes 'malloc.h' under Windows
```
#ifdef _WIN32
# include <malloc.h>
#endif
```

- 'malloc.h' is not available in CLANG (and not necessary, since its functions are available in 'stdlib.h' (already included by open62541 sources)

- I suggest to modify the above lines to
```
#ifdef _WIN32
# ifndef __clang__
#  include <malloc.h>
# endif
#endif
```
so that the library can be compiled with CLANG under Windows OS

jpfr, 2016-10-17:
Fixed in dfc313f7bd570cb7b7c68ba869eb2c893d11911e

closed 2016-10-17 by jpfr