open62541/open62541 #844: Add userland ability to not instantiate nodes

ichrispa, 2016-10-11:
Hi all,

the latest couple of issues like #836, #835 lead to an interesting problem. The XML file describes a snapshot of an exitent node hierarchy. When we create information modells in this way, we do not want to instantiate nodes - they are already instantiated by the model. Modellers like UAModeller infact create an instance tree automatically when a type is selected.

Further more, when we try to instantiate a model component, we usually muck up: the nodeIds are fixed by the model, but we choose our own. 

In short, we might end up with duplicate DataType/ObjectType structures. Or worse, creating the model might fail all together.

I would very much like to have a way to keep the stack from instantiating types when informationmodels are created. To that end, I would propose adding a switch to AddNode_existing: A boolean value inidicating that no instantiation should take place. 

I would append two functions to the highlevel API: addObjectNode_noInstantiate() and addVariableNode_noInstantiate(). Those would then become the default creation functions for i.e. the namespace compiler.

@jpfr, @markusgraube, @Stasik0: Would that be a viable option? Any counter-proposals on how to solve this problem?

Best regards,

ichrispa

jpfr, 2016-10-11:
You are right. This requires a good solution that sets the case once and for all.

addObjectNode_noInstantiate is fine for me.
In addition, I propose we also expose its complement, addObjectNode_finalize, to finish the instantiation, call the constructor callback, ... at a later time.

With your code that matches children by their browseName, doing the instantiation at a later time (after some children have been added manually) should be unproblematic.

Stasik0, 2016-10-12:
+1 for noInstantiate

markusgraube, 2016-10-12:
+1 this seems to be a good solution

ichrispa, 2016-10-17:
I saw a branche pop up... is anyone actively working on this?

FlorianPalm, 2016-10-17:
To get my demo working with the current master, I have added flag to distinguish if an instance should be created not ( addObjectNode_noInstantiate(), addVariableNode_noInstantiate()). Seems to work so far. I am going to clean it and create a pull request.

Pro, 2018-02-01:
There are now the `_begin`and `_finish` methods. See referenced issues

closed 2018-02-01 by Pro