open62541/open62541 #845: Remove #define NULL from namespace compiler

Pro, 2016-10-11:
@ichrispa  currently, the namespace compiler adds
`#define NULL ((void *)0)`

https://github.com/open62541/open62541/blob/0.2/tools/pyUANamespace/ua_namespace.py#L655

Is this really necessary?

It causes the compiler to throw warnings because NULL is already defined in string.h which is included in ua_config.h

ichrispa, 2016-10-11:
hmmm... I vaguely remember adding it for a reason. There was a construct UA_NULL, which was then removed or replaced for NULL. After that, some Compiler did not want to run properly without having NULL defined.

What about just wrapping it in ifndef?

ichrispa, 2016-10-11:
002cecd should fix the warnings for you @Pro.

Pro, 2016-10-11:
@ichrispa i think it's better if it is after the `#include "open62541.h"`

So change
```
header.append('  #ifndef NULL')
header.append('    #define NULL ((void *)0)')
header.append('  #endif NULL')
header.append('  #include "open62541.h"')
```
to
```
header.append('  #include "open62541.h"')
header.append('  #ifndef NULL')
header.append('    #define NULL ((void *)0)')
header.append('  #endif NULL')
```

Otherwise this definition of NULL may collide with some other definitions in open62541.h and its included headers

ichrispa, 2016-10-11:
Done :)

closed 2016-10-11 by ichrispa