open62541/open62541 #846: License Change (LGPL+Static Linking Exception -> Mozilla Public License v2)

FlorianPalm, 2016-10-11:
We are planning to change the license of open62541 project. It should be licensed under the Mozilla Public license v2 instead of our old LGPL with static linking exception. The MPLv2 license offers a broader vantage than the LGPL, as it is a standard license, thus legally easier to accept and more compatible with the juristic matters. Our intentions of changing our licensing is to expand and have a wider acceptance especially from corporates.

jpfr, 2016-10-17:
Great to see this happening!
MPLv2 is nearly exactly LGPL + static linking but in a nonmodified license.

We have consent in the core-team. That's 95% of the code covered...

Two questions regarding this.
- Is there a plan if one of the past contributors voices concerns?
- How long to wait until the transition?

HolgerJeromin, 2016-10-17:
I have not much code contributed but I am fine with changing the license.

Pro, 2017-03-08:
@FlorianPalm since the license is already changed in all the files should we just close this issue?

FlorianPalm, 2017-03-08:
yes : )

closed 2017-03-08 by FlorianPalm

bodorjzsolt, 2019-06-14:
In version 0.3-rc2 there are still some files with the old reference, pointing towards LGPL:
backend_open62541_nodes.py	
backend_open62541.py
datatypes.py
nodeset.py
nodes.py
Could you please correct this? Just by deleting the LGPL from the existing reference, should point now to the MPL-2.0.

rahulmohang, 2019-08-12:
Hello  Florian, 

Right now, can we consider the files, backend_open62541_nodes.py
backend_open62541.py
datatypes.py
nodeset.py
nodes.py

as MPL-2.0?

Pro, 2019-08-12:
@rahulmohang yes. Could you plase propose a PR which changes the license there? Thanks!

rahulmohang, 2019-08-12:
@Pro Thank you very much. Would have loved to do so. But, I am new to this, and do not know how to do so :-).