open62541/open62541 #847: Status Code to error message

Pro, 2016-10-12:
Is there already something like `strerror(...)` for UA_StatusCode?

Currently I only output the number, e.g. `0x80010000` but it would be nice to print an error message like:
`BadUnexpectedError: An unexpected error occurred (0x80010000)`

For that we need a global method which converts error numbers to strings, take from https://opcfoundation.org/UA/schemas/Opc.Ua.StatusCodes.csv

If it is not already there I would suggest something like this
https://opensource.apple.com/source/gcc/gcc-926/libiberty/strerror.c (it is public domain)
and a compiler switch to disable/enable the error table.

Any suggestions for the method name?
- UA_Strerrno(statuscode) and UA_Strerror
- UA_StatusCode_Msg and UA_StatusCode_Name

Should I go on with it and implement it?

jpfr, 2016-10-12:
Sounds good.
I propose returning static char strings like
```c
const char * UA_StatusCode_msg(UA_StatusCode code);
const char * UA_StatusCode_name(UA_StatusCode code);
```
That makes it easy to use the names in logging.

Writing a small generator script for the csv file might save time in the long run.
See also [/tools/generate_nodeids.py](https://github.com/open62541/open62541/blob/0.2/tools/generate_nodeids.py).

Stasik0, 2016-10-12:
But please make it configurable via cmake since it will blast the binary size...

Pro, 2016-10-12:
Yep, that was my intention, because it will definitely increase the binary size and not everyone needs it

Pro, 2016-10-13:
@jpfr should there also be a fallback version.

E.g. if a developer writes in his high-level code:
```
printf("Client connect failed: %s - %s", UA_StatusCode_name(retval), UA_StatusCode_msg(retval));
```
and open62541 isn't compiled with `UA_ENABLE_STATUSCODE_MSG` this will fail. But maybe the developer wants to support both versions, i.e. he has to write:

```
#ifdef UA_ENABLE_STATUSCODE_MSG
	    printf("Client connect failed: %s - %s", UA_StatusCode_name(retval), UA_StatusCode_msg(retval));
#endif
```

Would it be better to define a method which just returns the error code if `UA_ENABLE_STATUSCODE_MSG=OFF`

I.e:

```
#ifndef UA_ENABLE_STATUSCODE_MSG
char *code[11];
const char *UA_StatusCode_name(UA_StatusCode s) {
    sprintf(code, "0x%08x", s);
    return code;
}
const char *UA_StatusCode_msg(UA_StatusCode s) {
    sprintf(code, "0x%08x", s);
    return code;
}
#endif

commit 4c9857e1 (2016-10-17): Add missing statuscodes from StatusCode.csv

Add conversion from UA_StatusCode to human-readable string message. Fix #847

Add unit test for UA_StatusCode_msg and add to example

Add Opc.Ua.StatusCodes.csv for UA_StatusCode_msg
closed 2016-10-17 by jpfr