open62541/open62541 #848: 1.how UA_String translate to char array 2.identifierType Numeric or String?

xty19, 2016-10-12:
- 1. First

I have to translate UA_String to Char or String in C ++ code, i read the UA_String formate .
UA_String   data length,  i get the data, but there is somethings wrong with it .
For example:
`UA_String value = UA_STRING_ALLOC("iphone6s");

printf("value is  |%s|",value.data);`

->**value is |iphone6s?{@>|**.
and why?

- 2.Second 

some of my variable's **identifier** defined by **numeric** and some of them by **String** .
but in a function, i have to define a interface to call it as a common.
For example:
`void OPCOperator::OPC_Write_String(int opc_namespace, const char opc_nodeid[], char _value[]);
void OPCOperator::OPC_Write_String(int opc_namespace, int opc_nodeid, char _value[]);`
you know what i have to **implement the two function**, 
so how to just define only **one function** ?

so thank you for your sincerely help

caspervanzoest, 2016-10-12:
**1.** Your conversion from string to UA_String is good,
however, when you convert the UA_string into a string (with printf),
you need a better conversion.

Try this:
```
std::string l_nodeIdString = std::string((const char*)p_nodeid.identifier.string.data);
l_nodeIdString = l_nodeIdString.substr(0, p_nodeid.identifier.string.length);`
```
You can also do this with char array if you prefer.

**2.** For this issue I would implement a defaul paramenter, like this:

```
OPCOperator.h:

OPC_Write_String(int opc_namespace, const char opc_nodeid[], char _value[], int nodeIDNr = 0);
```
```
OPCOperator.cpp:

OPCOperator::OPC_Write_String(int opc_namespace, const char opc_nodeid[], char _value[], int nodeIDNr = 0)
{
    ...
    if(nodeIDNr == 0)
    {
        // add the node with string id
    }
    else
    {
         // add the node with int id
    }
    ...
}
```

xty19, 2016-10-13:
@caspervanzoest thank you so much.sorry for not replying on time .
**solution 2**. i think it is a good solution!:+1: 

**solution 1** i am testing ,i will reply latter.  :shipit:

xty19, 2016-10-13:
@caspervanzoest  thanks again 
**solution 1** work!
i have another question .
 if i read or write **some variable** ,
but variable have **many types**, 
such as UA_String UA_Init32 ,
the **method interface** how to define, or just **define many methods** to coordinate ?

caspervanzoest, 2016-10-13:
The typeis defined in UA_VariableTypeAttributes.

```
/**
 * VariableTypeAttributes
 * ----------------------
 * The attributes for a variable type node. */
typedef struct {
    UA_UInt32 specifiedAttributes;
    UA_LocalizedText displayName;
    UA_LocalizedText description;
    UA_UInt32 writeMask;
    UA_UInt32 userWriteMask;
    UA_Variant value;
    UA_NodeId dataType;
    UA_Int32 valueRank;
    size_t arrayDimensionsSize;
    UA_UInt32 *arrayDimensions;
    UA_Boolean isAbstract;
} UA_VariableTypeAttributes;
```

You could use a different method to set attributes foreach node.
Example:
```
UA_VariableAttributes OPCOperator::getAttributes(char[] type, char[] value)
{
	UA_VariableAttributes l_attr;
	UA_VariableAttributes_init(&l_attr);

	if (type == "UInt")
	{
		UA_UInt32 l_myVar= stoi(value);
		UA_Variant_setScalarCopy(&l_attr.value, &l_myVar, &UA_TYPES[UA_TYPES_UINT32]);
	}
	else if (type == "Bool")
	{
		UA_Boolean l_myVar = stoi(value);
		UA_Variant_setScalarCopy(&l_attr.value, &l_myVar, &UA_TYPES[UA_TYPES_BOOLEAN]);
	}
	else
	{
		printf("[ERROR] in OPCOperator.getAttributes : unknown dataType: %s\n", type);
	}
	return l_attr;
}
```

But anything is possible here. So its all up to you.
Good luck!

xty19, 2016-10-14:
@caspervanzoest  thanks you so much, i learn a lot.  i will work hard and do a lot of work on OPC-UA!
:+1:

closed 2016-10-14 by xty19