open62541/open62541 #852: xml transform to c code cost nearly 5 min

xty19, 2016-10-14:
just now ,i use Ua_Modeler to generate a xml which contains 3 variables :**UA_Int32 UA_Double UA_String**
and then i use it to generate c code,  it cost me **5 minutes**
 and what does it cost for everyone?

Pro, 2016-10-14:
Do you use CMAKE_BUILD_TYPE=Release?
This enabled code optimization which takes that long. Try to build with Debug.
See http://open62541.org/doc/current/tutorial_noderelations.html for further details, especially:

> 
After adding your XML file to CMakeLists.txt, rerun cmake in your build directory and enable DUA_ENABLE_GENERATE_NAMESPACE0. Make especially sure that you are using the option CMAKE_BUILD_TYPE=Debug. The generated namespace contains more than 30000 lines of code and many strings. Optimizing this amount of code with -O2 or -Os options will require several hours on most PCs! Also make sure to enable -DUA_ENABLE_METHODCALLS, as namespace 0 does contain methods that need to be encoded

Also note that there is an improved namespace compiler in #739

closed 2016-10-14 by Pro

xty19, 2016-10-14:
@Pro thanks , i will check it now :smile:

xty19, 2016-10-24:
@Pro it still cost 5 min ?
root@raspberrypi:/home/pi/open5/open62541/build# cmake -DCMAKE_BUILD_TYPE=Debug -DUA_ENABLE_METHODCALLS=On -BUILD_EXAMPLECLIENT=On -BUILD_EXAMPLESERVER=On -DUA_ENABLE_GENERATE_NAMESPACE0=On ..
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
-- Git failed (not a repo, or no tags). Build will not contain git revision info.
-- Git version: commit id unknown
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/open5/open62541/build
root@raspberrypi:/home/pi/open5/open62541/build# cmake ..
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
-- Git failed (not a repo, or no tags). Build will not contain git revision info.
-- Git version: commit id unknown
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/open5/open62541/build
root@raspberrypi:/home/pi/open5/open62541/build# make
[ 12%] Built target generate_informationmodel
[100%] Built target open62541-object
[100%] Built target open62541
root@raspberrypi:/home/pi/open5/open62541/build# 

root@raspberrypi:/home/pi/open5/open62541/tests/example_project_nomethod# /home/pi/open5/open62541/tools/pyUANamespace/generate_open62541CCode.py -i /home/pi/open5/open62541/tools/pyUANamespace/NodeID_Blacklist_FullNS0.txt /home/pi/open/open62541/tools/schema/namespace0/Opc.Ua.NodeSet2.xml example_project1024.xml example_project1024
INFO:__main__:Preprocessing /home/pi/open/open62541/tools/schema/namespace0/Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing example_project1024.xml
INFO:__main__:Parsing /tmp/Opc.Ua.NodeSet2.xml_preProcessed-p9tmIp.xml
INFO:__main__:Parsing /tmp/example_project1024.xml_preProcessed-ogFbqp.xml
INFO:__main__:Linking namespace nodes and references
INFO:__main__:Building datatype encoding rules
INFO:__main__:Allocating variables
INFO:__main__:Generating Header
INFO:__main__:Namespace generation code successfully printed
root@raspberrypi:/home/pi/open5/open62541/tests/example_project_nomethod#

Pro, 2016-10-24:
Ah, because you are compiling on a RspberryPi which is quite slow compared to a normal PC.
If you already use https://github.com/open62541/open62541/pull/739 then you can't do anything else here than wait.
Another solution would be to generate the code from XML on your PC and then upload it to the RasPI.

xty19, 2016-10-24:
:cry: Oh,  it really cost a lot of time for using raspberry
thanks for answer

Stasik0, 2016-10-25:
5 min ona  rpi isn't that bad, actually

xty19, 2016-10-26:
@Stasik0  OK,I WILL try it on my PC