open62541/open62541 #853: Chunking breaks client message assembly

ichrispa, 2016-10-14:
Hi all,

what exactly has been happening to our chunking? I am wondering why I get "Async not supported" errors on simple read requests, until wireshark spit this one out:

[image]

Apparently, chunking collates several messages into one TCP Frame. We then receive multiple frames from the server packed into a tcp frame - but they are appear to be completely out of order.

Is this packing multiple messages into one frame a feature or does it happen by mistake? Could we revert to the simpler yet more reliable one-request-per-message scheme?

Best regards,

ichrispa

Pro, 2016-10-14:
This may be related to #851, especially commit https://github.com/open62541/open62541/commit/3ce335cf221938037487cc63d3030a031772d1f3 by @jpfr?

Pro, 2016-10-14:
@ichrispa you could check if this also happend before the mentioned commit

ichrispa, 2016-10-14:
Can confirm without checking that it did not happen before #852

@jpfr: Please please check if you can fix this please... this bug is causing a minor amount of havoc in my autobuilds of half a dozen projects...

jpfr, 2016-10-14:
Hey @ichrispa ,

can you upload that wireshark dump of yours?
That would help in fixing and adding a unit test.
Thanks.

ichrispa, 2016-10-14:
Hi @jpfr:

I have been trying to reproduce this error to provide a clean debugging trace and a package dump... and I can't! It occured when running a 0.2 server against a master built client in a multithreaded application.
There still is an issue related to this one. In a multithreaded application, I sometimes do manage to send two requests A and B one directly after the other, with A then receiving the request for B. Clearly, we do not support async communications and this error is on me. But there is also no possible way to recover from this error.
With the more generic 'processChunk()' routine and its callback, there is no easy way to discard the package and advance the requestId to the new value. Hence we get stuck with wrong requestId in the client and no serverside message matches the clients decoder anymore. 

I would just chalk this one up with "we need asynchronous communications at some point".

Sorry for the fuss @jpfr, this one is entirely my fault. Thanks you for your quick reply non the less.

closed 2016-10-14 by ichrispa