open62541/open62541 #859: Variables from ns0 in typedefinitions missing

j-sc, 2016-10-19:
Some variables (e.g. i=2769) from namespace0 in _Opc.Ua.NodeSet2.Optimal.xml_ that are components of Objecttypes are missing in the server. They are in _ua_namespaceinit_generated.c_ so it's probably not the namespace-compiler.

j-sc, 2016-10-21:
I made a minimal example showing the problem. Both _TestVariable1_ and _TestVariable2_ should be a component of _TestObjectType_ but only _TestVariable2_ is created.
[server.txt](https://github.com/open62541/open62541/files/543668/server.txt)

Pro, 2016-10-21:
@j-sc it's better if you directly post the code in the code tags, like this:

```
/* This work is licensed under a Creative Commons CCZero 1.0 Universal License.
 * See http://creativecommons.org/publicdomain/zero/1.0/ for more information. */

#include <signal.h>

#ifdef UA_NO_AMALGAMATION
#include "ua_types.h"
#include "ua_server.h"
#include "ua_config_standard.h"
#include "ua_network_tcp.h"
#else
#include "open62541.h"
#endif

UA_Boolean running = true;
static void stopHandler(int sig) {
    running = false;
}

int main(void) {
    signal(SIGINT,  stopHandler);
    signal(SIGTERM, stopHandler);

    UA_ServerConfig config = UA_ServerConfig_standard;
    UA_ServerNetworkLayer nl = UA_ServerNetworkLayerTCP(UA_ConnectionConfig_standard, 16664);
    config.networkLayers = &nl;
    config.networkLayersSize = 1;
    UA_Server *server = UA_Server_new(config);
    
    /* Create a rudimentary objectType
     * 
     * BaseObjectType
     * |
     * +- (OT) TestObjectType
     *    + (V) TestVariable1 <TestVariableType>
     *    + (V) TestVariable2 <BaseDataVariableType>
     */ 
    UA_StatusCode retval;
    UA_ObjectTypeAttributes otAttr;
    UA_ObjectTypeAttributes_init(&otAttr);
    otAttr.displayName = UA_LOCALIZEDTEXT("en_US", "TestObjectType");
		otAttr.description = UA_LOCALIZEDTEXT("", "");
    UA_Server_addObjectTypeNode(server, UA_NODEID_NUMERIC(1, 1000), 
                                UA_NODEID_NUMERIC(0, UA_NS0ID_BASEOBJECTTYPE), UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE),
                                UA_QUALIFIEDNAME(1, "TestObjectType"), otAttr, NULL, NULL);

    UA_VariableTypeAttributes vtAttr;
		UA_VariableTypeAttributes_init(&vtAttr);
		vtAttr.displayName = UA_LOCALIZEDTEXT("", "TestVariableType");
		vtAttr.description = UA_LOCALIZEDTEXT("", "");
		vtAttr.valueRank = -1;
		UA_NodeId nodeId = UA_NODEID_NUMERIC(1, 2000);
		UA_NodeId typeDefinition = UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE);
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, 63);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, 45);
    UA_QualifiedName nodeName = UA_QUALIFIEDNAME(0, "StateVariableType");
    UA_Server_addVariableTypeNode(server, nodeId, parentNodeId, parentReferenceNodeId, nodeName,
																	typeDefinition, vtAttr, NULL, NULL);

    UA_VariableAttributes   vAttr;
    UA_VariableAttributes_init(&vAttr);
    vAttr.displayName =  UA_LOCALIZEDTEXT("en_US", "TestVariable1");
		vAttr.description = UA_LOCALIZEDTEXT("", "");
    UA_UInt32 testVar = 3;
    UA_Variant_setScalarCopy(&vAttr.value, &testVar, &UA_TYPES[UA_TYPES_UINT32]);
    UA_Server_addVariableNode(server, UA_NODEID_NUMERIC(1, 1001), 
                              UA_NODEID_NUMERIC(1, 1000), UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
                              UA_QUALIFIEDNAME(1, "TestVariable1"), UA_NODEID_NUMERIC(1, 2000), vAttr, NULL, NULL);

		UA_VariableAttributes   vAttr2;
    UA_VariableAttributes_init(&vAttr2);
    vAttr2.displayName =  UA_LOCALIZEDTEXT("en_US", "TestVariable2");
		vAttr2.description = UA_LOCALIZEDTEXT("", "");
    UA_UInt32 testVar2 = 3;
    UA_Variant_setScalarCopy(&vAttr2.value, &testVar2, &UA_TYPES[UA_TYPES_UINT32]);
    UA_Server_addVariableNode(server, UA_NODEID_NUMERIC(1, 1002), 
                              UA_NODEID_NUMERIC(1, 1000), UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
                              UA_QUALIFIEDNAME(1, "TestVariable2"), UA_NODEID_NUMERIC(0, 63), vAttr2, NULL, NULL);
  
    retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    nl.deleteMembers(&nl);
    return (int)retval;
}
```

You should check the return values of `UA_Server_add...Node`. It should at least give you more information why the node is not created.

If you do so, you can see that the first call to `UA_Server_addVariableNode` fails with:

> [10/21/2016 07:18:38.887] debug/session	Connection 0 | SecureChannel 0 | Session {00000001-0000-0000-0000-000000000000} | AddNodes: Could not prepare the new node with status code 0x80740000

> Error: The value supplied for the attribute is not of the same type as the attribute's value.

Maybe this helps you debugging the problem?

j-sc, 2016-10-21:
thanks for the hint on where to look. I think I found a solution.
In `ua_services_attribute.c` changing the `valueRank` to -1 wasn't permitted.

```
 /* Check if the valuerank of the type allows the change */
    switch(vt->valueRank) {
    case -3: /* the value can be a scalar or a one dimensional array */
        if(valueRank != -1 && valueRank != 1)
            return UA_STATUSCODE_BADTYPEMISMATCH;
        break;
    case -2: /* the value can be a scalar or an array with any number of dimensions */
       	break;
    case -1: /* the value is a scalar */
      	return UA_STATUSCODE_BADTYPEMISMATCH;
	////// should be
        //if(valueRank != -1)
        //    return UA_STATUSCODE_BADTYPEMISMATCH;
        //break;
    case 0: /* the value is an array with one or more dimensions */
        if(valueRank < 0)
            return UA_STATUSCODE_BADTYPEMISMATCH;
        break;
    default: /* >= 1: the value is an array with the specified number of dimensions */
        return UA_STATUSCODE_BADTYPEMISMATCH;
    }
```
With the change in the comment all of my variableNodes are created.

Pro, 2016-10-21:
@jpfr I also think that this additional if should be included.
I.e., if `vt` has valueRank=-1 and the new value rank for the node is also -1 then it should be allowed to set the value rank?

jpfr, 2016-10-24:
Right. Implemented in bf4e481.
Does this fix the issue @j-sc ?

j-sc, 2016-10-26:
No, it actually made it worse. If I only change the check of the valueRank it works for me. Changing only the check shouldn't break anything else.

jpfr, 2016-10-27:
The change was revamped.
Can you test with b6ff129f5f594e10d73bb202867c74a1f3f1b60b or provide a non-working example?
The unit tests are all green.

j-sc, 2016-10-27:
It works.

closed 2016-10-27 by jpfr