open62541/open62541 #861: Stress testing question

StalderT, 2016-10-23:
Hello,

I have add to server.c :

```
static UA_StatusCode
readTest(void *handle, const UA_NodeId nodeId, UA_Boolean sourceTimeStamp,
             const UA_NumericRange *range, UA_DataValue *value) {
    if(range) {
        value->hasStatus = UA_TRUE;
        value->status = UA_STATUSCODE_BADINDEXRANGEINVALID;
        return UA_STATUSCODE_GOOD;
    }   
    UA_UInt32 val = (UA_UInt32)handle;
    UA_DateTime currentTime = UA_DateTime_now();
    UA_Variant_setScalarCopy(&value->value, &val, &UA_TYPES[UA_TYPES_UINT32]);
    value->hasValue = UA_TRUE;
    if(sourceTimeStamp) {
        value->hasSourceTimestamp = UA_TRUE;
        value->sourceTimestamp = currentTime;
    }   
    return UA_STATUSCODE_GOOD;
}
```
and
```
  for(UA_Uint32 i=0; i<10000; i++)
    {
            UA_DataSource dateDataSource_test = (UA_DataSource) {.handle = (void*)i, .read = readTest, .write = NULL};
            UA_VariableAttributes v_attr_test;
            UA_VariableAttributes_init(&v_attr_test);
            char name[100];
            snprintf(name, 100, "TEST%04d", i);
            v_attr_test.description = UA_LOCALIZEDTEXT("en_US",name);
            v_attr_test.displayName = UA_LOCALIZEDTEXT("en_US",name);
            const UA_QualifiedName dateName_test = UA_QUALIFIEDNAME(1, name);
            UA_NodeId dataSourceId_test;
            UA_Server_addDataSourceVariableNode(server, UA_NODEID_NULL,
                                                UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                                                UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES), dateName_test,
                                                UA_NODEID_NULL, v_attr_test, dateDataSource_test, &dataSourceId_test);
    }
```
If I try to add 10000 tests variables to Data Access View in UaExpert server use 100% CPU.

If I try with https://github.com/open62541/open62541/commit/94c1070fc815f7363bc75f240a93fb19d1c74dbb all works fine (see https://github.com/open62541/open62541/issues/541)

I would like to know if the implementation has changed or if it is a bug in the current master version ?

jpfr, 2016-10-27:
Fixed in #868.

We had a speed regression in the worker code.
Every repeated job did a linear search for the next position in the sorted list of repeated jobs.
That lead to a quadratic increase in runtime in the number of repeated jobs.

I see 20% CPU consumption for 10.000 subscriptions on my laptop.

closed 2016-10-27 by jpfr

StalderT, 2016-10-31:
@jpfr 

If I create 10.000 subscriptions, 10.000 RepeatedJob are created.

It consume many CPU time. I have made some tests :

1) Change 
```
void UA_MoniteredItem_SampleCallback(UA_Server *server, UA_MonitoredItem *monitoredItem) {
    UA_Subscription *sub = monitoredItem->subscription;
    if(monitoredItem->monitoredItemType != UA_MONITOREDITEMTYPE_CHANGENOTIFY) {
        UA_LOG_DEBUG_SESSION(server->config.logger, sub->session,
                             "Subscription %u | MonitoredItem %i | "
                             "Not a data change notification",
                             sub->subscriptionID, monitoredItem->itemId);
        return;
    }   
```
to
```
void UA_MoniteredItem_SampleCallback(UA_Server *server, UA_MonitoredItem *monitoredItem) {
    UA_Subscription *sub = monitoredItem->subscription;
    if(monitoredItem->monitoredItemType != UA_MONITOREDITEMTYPE_CHANGENOTIFY) {
        UA_LOG_DEBUG_SESSION(server->config.logger, sub->session,
                             "Subscription %u | MonitoredItem %i | "
                             "Not a data change notification",
                             sub->subscriptionID, monitoredItem->itemId);
        return;
    }   
    return; // DO NOTHING
```
Result : High CPU time.

2) Change
```
UA_StatusCode
MonitoredItem_registerSampleJob(UA_Server *server, UA_MonitoredItem *mon) {
    UA_Job job;
    job.type = UA_JOBTYPE_METHODCALL;
    job.job.methodCall.method = (UA_ServerCallback)UA_MoniteredItem_SampleCallback;
    job.job.methodCall.data = mon;
    UA_StatusCode retval = UA_Server_addRepeatedJob(server, job,
                                                    (UA_UInt32)mon->samplingInterval,
                                                    &mon->sampleJobGuid);
    if(retval == UA_STATUSCODE_GOOD)
        mon->sampleJobIsRegistered = true;
    return retval;
}
```
to 
```
UA_StatusCode
MonitoredItem_registerSampleJob(UA_Server *server, UA_MonitoredItem *mon) {
#if 0
    UA_Job job;
    job.type = UA_JOBTYPE_METHODCALL;
    job.job.methodCall.method = (UA_ServerCallback)UA_MoniteredItem_SampleCallback;
    job.job.methodCall.data = mon;
    UA_StatusCode retval = UA_Server_addRepeatedJob(server, job,
                                                    (UA_UInt32)mon->samplingInterval,
                                                    &mon->sampleJobGuid);
#else
    UA_StatusCode retval = UA_STATUSCODE_GOOD;
#endif

    if(retval == UA_STATUSCODE_GOOD)
        mon->sampleJobIsRegistered = true;
    return retval;
}
```
Result: Low CPU time

This suggests that it is the RepeatedJob treatment who consumes CPU ressoures. Would it make sense to review the management of SampleCallback?

One solution might be to create a RepeatedJob for every unique/single samplingInterval. For example, if the client creates 1.000 subscriptions (with 100ms samplingInterval),  only 1 RepeatedJob will be created and this RepeatedJob  will call 1.000 SampleCallback.

I hope you will find this understandable. Sorry for my bad English.

jpfr, 2016-10-31:
> This suggests that it is the RepeatedJob treatment who consumes CPU ressoures. Would it make sense to review the management of SampleCallback?

Well of course. Without the repeatedjob, we never sample values to detect changes. And without changes, we never send a notification.

> One solution might be to create a RepeatedJob for every unique/single samplingInterval

That is quite possible. A second way is to add a counter value to every node that is increased for every change. But we need to get around that for data sources.

> I hope you will find this understandable. Sorry for my bad English.

Your written english is quite well. 😄 

We had quite some improvements in CPU-time for subscriptions. I can do 10.000 monitoreditems on my laptop. What is your current CPU consumption and what is your target?
We need to plan how we spend resources. Because there are many things we could optimize/implement instead.

StalderT, 2016-10-31:
-- This suggests that it is the RepeatedJob treatment who consumes CPU ressoures. Would it make sense to review the management of SampleCallback?
-- Well of course. Without the repeatedjob, we never sample values to detect changes. And without changes, we never send a notification.
Ok, but in the first case, we never sample values and the CPU usage is very high.

My target are on a VM (one i7 CPU/CORE)

It seems that it is this loop which consumes a lot of CPU:

```
        /* Keep the list sorted */
        struct RepeatedJob *prev_rj = lastNow;
        while(true) {
            struct RepeatedJob *n = LIST_NEXT(prev_rj, next);
            if(!n || n->nextTime >= rj->nextTime)
                break;
            prev_rj = n;
        }
        if(prev_rj != rj) {
            LIST_REMOVE(rj, next);
            LIST_INSERT_AFTER(prev_rj, rj, next);
        }

```

Without this loop : 6% CPU
With this loop : 90% CPU

But I understand, that we can not remove the list sortering.

StalderT, 2016-11-01:
And something like that : https://github.com/open62541/open62541/compare/0.2...StalderT:patch-5

A processRepeatedJobs without "Keep the list sorted".  "next" variable are tested for each list item.

What do you think?

StalderT, 2016-11-02:
Sorry, my proposale dont work. Please ignore it.