open62541/open62541 #862: UaModeler variable :BadWaitingForInitialData?

xty19, 2016-10-24:
i use xml to generate c code .
but when i use UaExpert to see the variables . 
none of them can be seen.
Statuscode : BadWaitingForInitialData
why
i just use the cmake command:
cmake -DCMAKE_BUILD_TYPE=Debug -DUA_ENABLE_METHODCALLS=On \
        -BUILD_EXAMPLECLIENT=On -BUILD_EXAMPLESERVER=On \
        -DUA_ENABLE_GENERATE_NAMESPACE0=On ../

xty19, 2016-10-24:
@Pro 
@jpfr 
i dont know why?
because it can work just now just after i use the cmake command above.

caspervanzoest, 2016-10-25:
Do you mean you are using XML to generate the nodes?
Have you tried those examples:
[https://github.com/open62541/open62541/blob/master/examples/server_nodeset.xml](https://github.com/open62541/open62541/blob/master/examples/server_nodeset.xml)
[https://github.com/open62541/open62541/blob/master/examples/server_nodeset.c](https://github.com/open62541/open62541/blob/master/examples/server_nodeset.c)

Let me know if you need a look at your code.

xty19, 2016-10-25:
@caspervanzoest 
thanks i made it.
just because the TypeDefinition ,
i choose the **BaseVariableType** instead of **Property Type**  and solve the problem

closed 2016-10-25 by xty19

caspervanzoest, 2016-10-25:
Nice.

Could you do me a big favor, by helping me on issue [#864](https://github.com/open62541/open62541/issues/864)

Could you try the Prosys client on your server, and tell me if you are able to connect?

Thank you in advance!