open62541/open62541 #863: Overriding method from base type

Pro, 2016-10-24:
I have a conceptual question about OPC UA and open62541:
Is, or should it be possible, to overwrite object methods in an instantiated object?

Let's say I create a new ObjectTypeNode as a base object:

```
- SkillBaseType
   - Variable: Description(String)
   - Method: InvokeSkill
      - InputArguments: input(BaseDataType)
      - OutputArguments: output(BaseDataType)
```

Now I want to create an instance of the SkillBaseType, but overwrite the method arguments:

```
      - InputArguments: x(Int32)
      - OutputArguments: x_sqrt(Double)
```

I tried it with the following code, but then I see two methods with the same BrowseName and the one from the base is not overwritten.

Base Type:
```
	UA_ObjectTypeAttributes otAttr;
	UA_ObjectTypeAttributes_init(&otAttr);

	otAttr.description = UA_LOCALIZEDTEXT("en_US", "Generic type for describing skill elements");
	otAttr.displayName = UA_LOCALIZEDTEXT("en_US", "SkillBaseType");
	UA_Server_addObjectTypeNode(msb->server, UA_NODEID_STRING(nsIdx, "SkillBaseType"),
								UA_NODEID_NUMERIC(0, UA_NS0ID_BASEOBJECTTYPE), UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE),
								UA_QUALIFIEDNAME(nsIdx, "SkillBaseType"), otAttr, NULL, NULL);

	UA_VariableAttributes deviceDescriptionAttr;
	UA_VariableAttributes_init(&deviceDescriptionAttr);
	deviceDescriptionAttr.description = UA_LOCALIZEDTEXT("en_US", "Device description of the skill");
	deviceDescriptionAttr.displayName = UA_LOCALIZEDTEXT("en_US", "DeviceDescription");
	UA_String deviceDescription = UA_String_fromChars("");
	UA_Variant_setScalarCopy(&deviceDescriptionAttr.value, &deviceDescription, &UA_TYPES[UA_TYPES_STRING]);
	UA_Server_addVariableNode(msb->server, UA_NODEID_STRING(nsIdx, "DeviceDescription"),
							  UA_NODEID_STRING(nsIdx, "SkillBaseType"), UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY),
							  UA_QUALIFIEDNAME(nsIdx, "DeviceDescription"), UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), deviceDescriptionAttr, NULL,
							  NULL);

	UA_Argument inputArguments1;
	UA_Argument_init(&inputArguments1);
	inputArguments1.dataType = UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATATYPE);
	inputArguments1.description = UA_LOCALIZEDTEXT("en_US", "Base Input");
	inputArguments1.name = UA_STRING("input");
	inputArguments1.valueRank = -2; // Any

	UA_Argument outputArguments1;
	UA_Argument_init(&outputArguments1);
	outputArguments1.dataType = UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATATYPE);
	outputArguments1.description = UA_LOCALIZEDTEXT("en_US", "Base Output");
	outputArguments1.name = UA_STRING("output");
	outputArguments1.valueRank = -2; // Any

	UA_MethodAttributes connect_skill_device_attr;
	UA_MethodAttributes_init(&connect_skill_device_attr);
	connect_skill_device_attr.description = UA_LOCALIZEDTEXT("en_US", "Triggers the underlying skill of the device.");
	connect_skill_device_attr.displayName = UA_LOCALIZEDTEXT("en_US", "InvokeSkill");
	connect_skill_device_attr.executable = true;
	connect_skill_device_attr.userExecutable = true;
	UA_Server_addMethodNode(server, UA_NODEID_STRING(nsIdx, "InvokeSkillBase"),
							UA_NODEID_STRING(nsIdx, "SkillBaseType"),
							UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
							UA_QUALIFIEDNAME(nsIdx, "InvokeSkill"),
							connect_skill_device_attr, &invokeSkill, NULL,
							1, &inputArguments1, 1, &outputArguments1, NULL);

```

And then the instantiation:

```
	UA_ObjectAttributes oAttr;
	UA_ObjectAttributes_init(&oAttr);
	oAttr.description = UA_LOCALIZEDTEXT("en_US", "A motor realizing generic skill type.");
	oAttr.displayName = UA_LOCALIZEDTEXT("en_US", "Skill");
	UA_StatusCode retVal;
	if ((retVal = UA_Server_addObjectNode(server, UA_NODEID_STRING(nsIdx, "Skill"),
							UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER), UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
							UA_QUALIFIEDNAME(nsIdx, "Skill"), UA_NODEID_STRING(nsIdx, "SkillBaseType"), oAttr, NULL, NULL)) != UA_STATUSCODE_GOOD) {
		UA_LOG_ERROR(logger, UA_LOGCATEGORY_SERVER, "Could not add Skill node: %s", UA_StatusCode_description(retVal));
	}

	UA_Argument inputArguments1;
	UA_Argument_init(&inputArguments1);
	inputArguments1.dataType = UA_TYPES[UA_TYPES_DOUBLE].typeId;
	inputArguments1.description = UA_LOCALIZEDTEXT("en_US", "A value");
	inputArguments1.name = UA_STRING("x");
	inputArguments1.valueRank = -1;

	UA_Argument outputArguments1;
	UA_Argument_init(&outputArguments1);
	outputArguments1.dataType = UA_TYPES[UA_TYPES_DOUBLE].typeId;
	outputArguments1.description = UA_LOCALIZEDTEXT("en_US", "The positive square root of x. If the argument is NaN or less than zero, the result is NaN.");
	outputArguments1.name = UA_STRING("x_sqrt");
	outputArguments1.valueRank = -1;

	UA_MethodAttributes connect_skill_device_attr;
	UA_MethodAttributes_init(&connect_skill_device_attr);
	connect_skill_device_attr.description = UA_LOCALIZEDTEXT("en_US","Triggers the underlying skill of the device.");
	connect_skill_device_attr.displayName = UA_LOCALIZEDTEXT("en_US","InvokeSkill");
	connect_skill_device_attr.executable = true;
	connect_skill_device_attr.userExecutable = true;
	UA_StatusCode retVal = UA_Server_addMethodNode(server, UA_NODEID_STRING(nsIdx, "InvokeSkill"),
												   UA_NODEID_STRING(nsIdx, "Skill"),
												 UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
												 UA_QUALIFIEDNAME(nsIdx, "InvokeSkill"),
												 connect_skill_device_attr, &invokeSkill, server,
												 1, &inputArguments1, 1, &outputArguments1, NULL);
```

[image]

Is this the correct way to do it? If yes, why does overriding the method not work?

jpfr, 2016-10-31:
Hey @Pro,

See #844 and #857. We are splitting the object instantiation into two parts.
1. Creating the node
2. Adding children and calling the constructor

@ichrispa provided some great functionality where we look for an existing member with the same browseName and do not instantiate.

So in the future, the steps would be:
Create the object (first part), add you method, finish the object instantiation (2nd part).
Then, the duplicate method will not be copied over from the object type.

Would that work for you?

Pro, 2016-10-31:
Sounds good! I think that's the correct way.

closed 2016-10-31 by Pro