open62541/open62541 #864: "Decoding of message header failed"

caspervanzoest, 2016-10-24:
I am making my OPC server using open62541,
and using clients from 'Prosys OPC'.
The client I am using: [https://downloads.prosysopc.com/opc-ua-client-downloads.php](url)

The server API seems to fail to accept incoming connections from the Prosys client.
even with [https://github.com/open62541/open62541/blob/0.2/examples/server.c](url)

However, once in +/- 20 attempts, the connection seems to be successful.

I managed to capture both connection succes and failure in Wireshark:

Any suggestion is welcome.
Thanks in advance.

Regards,

Casper

Pro, 2016-10-24:
@caspervanzoest the pcap files did not upload comletely.
Can you post them again?

caspervanzoest, 2016-10-24:
My bad, my organisations seems to block all uploads. 
Please try [https://drive.google.com/open?id=0BwpWtgvWFKg8ZVBPV0ZObnpCdnc](url)
Thanks!

jpfr, 2016-10-24:
Hey @caspervanzoest,

it seems your network is a bit unstable.
Here's a Wireshark screenshot of the not-working case.

[image]

The Hello message is split into three parts. The first packet contains the single character 'H' as payload.
After the HEL, the connection is a aborted with a TCP reset.
It seems the operating system is closing the connection.

If you need more help or really suspect our implementation to be faulty, please use a current version from the 0.2 branch and activate a log-level DEBUG or below to get more insight.

We can then take a closer look.

caspervanzoest, 2016-10-25:
I honestly do think the implementation might be faulty.
The quickest way to find out if my network is the bottleneck,
we should try the [prosysopc.com opc-ua-client](https://downloads.prosysopc.com/opc-ua-client-downloads.php) on a different network.

Could you please try the given client using any open62541 server?
If you have the ability to connect right away, We will know where the issue is.

Thanks alot!

Pro, 2016-10-25:
@caspervanzoest I just ran the opc-ua-client, and it worked without any problem.

OS: Ubuntu 16.04
open62541: most current 0.2 branch, examples/server_datasource.c
Prosys Client: just downloaded from your provided link, and installed as .deb

I also tried multiple times, no problems here. I could read the value of the variable node.

Do you have multiple network cards in your PC? You could try to disable all of them and only enable the main network card.

Otherwise try to use other tools which are able to measure the network quality.

Stasik0, 2016-10-25:
So what was the answer?

caspervanzoest, 2016-10-25:
My conclu

caspervanzoest, 2016-10-25:
My conclusion for now is, my network is unstable which causes connection issues. As pro did a proof of concept, and jpfr showed his point of view, I totally do agree with this conclusion.

closed 2016-10-25 by caspervanzoest