open62541/open62541 #865: a client close in 10min? how to maintain it?

xty19, 2016-10-25:
i make a server on raspberry. 
and make a client on my Pc.
Pc client  write variables on server.
i made it as an interface to call for convenience.
but when i call it after 10 min. the server console say " Colsing the Connection number".
 
what does your friends  solve this problem?

`	void OPCOperator::OPC_Init(const char opc_url[]);
void OPCOperator::OPC_Write_Int32(int opc_namespace, const char opc_nodeid[], int opc_nodeIDNr = 0,int value=0);`

just at the beginning of the program i initialize the OPC UA client.
and then i call it to write the variables.

Stasik0, 2016-10-25:
You need to call some client actions, e.g. read, otherwise server will close the connection. Doing this automatically in another thread is on the todo list.

xty19, 2016-10-25:
@Stasik0 
thanks for reply.
what does **on the todo list.** mean?
would you have some example code?
really appreciate your answer :+1:

xty19, 2016-10-26:
@Stasik0  
I read **server variable every minute** but it still close after 10 min :cry:

jpfr, 2016-10-26:
Hey @xty19 ,

set the log-level to DEBUG or below. Then you see a reason why the connection is closed.

Stasik0, 2016-10-26:
This shouldnt happen, we will need wireshark traces. Are you using master branch?

xty19, 2016-11-07:
@Stasik0 thanks you very much ,i will try wireshark traces,and i use the master branch.
@jpfr thanks , set the log-level to DEBUG or below. ?  how to add code ,would you plase say it in detail ,thanks again :smiley:

jpfr, 2016-11-07:
Hello @xty19 ,

> set the log-level to DEBUG or below. ? how to add code ,would you plase say it in detail ,thanks again 

Either in cmake or in the resulting open62541.h set UA_LOGLEVEL to a numeric value of 100 or below.

xty19, 2016-11-08:
@jpfr ok, i will try it . :smiley:

jpfr, 2016-11-28:
Hey @xty19,

is this still open?

xty19, 2016-11-29:
@jpfr  it can set the close time ,no matter you read or write some variables, you have to renew a connection. i use the code to maintain it. and it work . and the wine shark runs with some error in my computer environment.
`retval=UA_Client_connect(client, opc_url);
	if (retval != UA_STATUSCODE_GOOD)
	{
		UA_Client_delete(client);
		client = UA_Client_new(UA_ClientConfig_standard);
		retval = UA_Client_connect(client, opc_url);
		printf("\n OPC NEW\n");
	}`

closed 2016-11-29 by xty19