open62541/open62541 #869: source/server timestamp question/problem

StalderT, 2016-10-29:
Hello,

I have add to server.c :

```
static UA_StatusCode
readTest(void *handle, const UA_NodeId nodeId, UA_Boolean sourceTimeStamp,
             const UA_NumericRange *range, UA_DataValue *value) {
    if(range) {
        value->hasStatus = UA_TRUE;
        value->status = UA_STATUSCODE_BADINDEXRANGEINVALID;
        return UA_STATUSCODE_GOOD;
    }   
    UA_UInt32 val = (UA_UInt32)handle;
    UA_DateTime currentTime = UA_DateTime_now();
    UA_Variant_setScalarCopy(&value->value, &val, &UA_TYPES[UA_TYPES_UINT32]);
    value->hasValue = UA_TRUE;
    if(sourceTimeStamp) {
        value->hasSourceTimestamp = UA_TRUE;
        value->sourceTimestamp = currentTime;
    }   
    return UA_STATUSCODE_GOOD;
}
```
and
```
  for(UA_Uint32 i=0; i<10000; i++)
    {
            UA_DataSource dateDataSource_test = (UA_DataSource) {.handle = (void*)i, .read = readTest, .write = NULL};
            UA_VariableAttributes v_attr_test;
            UA_VariableAttributes_init(&v_attr_test);
            char name[100];
            snprintf(name, 100, "TEST%04d", i);
            v_attr_test.description = UA_LOCALIZEDTEXT("en_US",name);
            v_attr_test.displayName = UA_LOCALIZEDTEXT("en_US",name);
            const UA_QualifiedName dateName_test = UA_QUALIFIEDNAME(1, name);
            UA_NodeId dataSourceId_test;
            UA_Server_addDataSourceVariableNode(server, UA_NODEID_NULL,
                                                UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                                                UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES), dateName_test,
                                                UA_NODEID_NULL, v_attr_test, dateDataSource_test, &dataSourceId_test);
    }
```
If I monitor TEST0001 with UaExpert, source timestamp and server timestamp always change.

In detectValueChange function, UA_String_equal(encoding, &mon->lastSampledValue) is always equal to 0, "changed" is set to true and value always resend to the client.

How can we change the code to only send the value if the value realy change?

StalderT, 2016-10-29:
I don't know if I understand, but :

UA_encodeBinary in detectValueChange function encode 6 bytes (binsize for a UA_UInt32 node). After UA_String_equal compare "encoding" and "&mon->lastSampledValue" but both have a size of 512 bytes. And 512 bytes are compared, not only the 6 firsts bytes.

Is it correct?

StalderT, 2016-10-31:
Without the pr, with valgrind we have : 

```
==21690== Conditional jump or move depends on uninitialised value(s)
==21690==    at 0x4030104: bcmp (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==21690==    by 0x804D94E: UA_String_equal (in /data/a.out)
==21690==    by 0x80661F9: detectValueChange (in /data/a.out)
==21690==    by 0x80663BE: UA_MoniteredItem_SampleCallback (in /data/a.out)
==21690==    by 0x805BD17: processJob (in /data/a.out)
==21690==    by 0x805BFC4: processRepeatedJobs (in /data/a.out)
==21690==    by 0x805C46B: UA_Server_run_iterate (in /data/a.out)
==21690==    by 0x805C6EA: UA_Server_run (in /data/a.out)
==21690==    by 0x804BC99: main (in /data/a.out)
==21690== 
==21690== Conditional jump or move depends on uninitialised value(s)
==21690==    at 0x403013F: bcmp (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==21690==    by 0x804D94E: UA_String_equal (in /data/a.out)
==21690==    by 0x80661F9: detectValueChange (in /data/a.out)
==21690==    by 0x80663BE: UA_MoniteredItem_SampleCallback (in /data/a.out)
==21690==    by 0x805BD17: processJob (in /data/a.out)
==21690==    by 0x805BFC4: processRepeatedJobs (in /data/a.out)
==21690==    by 0x805C46B: UA_Server_run_iterate (in /data/a.out)
==21690==    by 0x805C6EA: UA_Server_run (in /data/a.out)
==21690==    by 0x804BC99: main (in /data/a.out)
==21690== 
==21690== Conditional jump or move depends on uninitialised value(s)
==21690==    at 0x8066202: detectValueChange (in /data/a.out)
==21690==    by 0x80663BE: UA_MoniteredItem_SampleCallback (in /data/a.out)
==21690==    by 0x805BD17: processJob (in /data/a.out)
==21690==    by 0x805BFC4: processRepeatedJobs (in /data/a.out)
==21690==    by 0x805C46B: UA_Server_run_iterate (in /data/a.out)
==21690==    by 0x805C6EA: UA_Server_run (in /data/a.out)
==21690==    by 0x804BC99: main (in /data/a.out)
==21690== 
==21690== Conditional jump or move depends on uninitialised value(s)
==21690==    at 0x4030124: bcmp (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==21690==    by 0x804D94E: UA_String_equal (in /data/a.out)
==21690==    by 0x80661F9: detectValueChange (in /data/a.out)
==21690==    by 0x80663BE: UA_MoniteredItem_SampleCallback (in /data/a.out)
==21690==    by 0x805BD17: processJob (in /data/a.out)
==21690==    by 0x805BFC4: processRepeatedJobs (in /data/a.out)
==21690==    by 0x805C46B: UA_Server_run_iterate (in /data/a.out)
==21690==    by 0x805C6EA: UA_Server_run (in /data/a.out)
==21690==    by 0x804BC99: main (in /data/a.out)
==21690== 
==21690== Conditional jump or move depends on uninitialised value(s)
==21690==    at 0x40300F5: bcmp (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==21690==    by 0x804D94E: UA_String_equal (in /data/a.out)
==21690==    by 0x80661F9: detectValueChange (in /data/a.out)
==21690==    by 0x80663BE: UA_MoniteredItem_SampleCallback (in /data/a.out)
==21690==    by 0x805BD17: processJob (in /data/a.out)
==21690==    by 0x805BFC4: processRepeatedJobs (in /data/a.out)
==21690==    by 0x805C46B: UA_Server_run_iterate (in /data/a.out)
==21690==    by 0x805C6EA: UA_Server_run (in /data/a.out)
==21690==    by 0x804BC99: main (in /data/a.out)
==21690== 
```

jpfr, 2016-10-31:
Fixed in 2a127dc9e361a29cc76428af80544e547262f3ee

closed 2016-10-31 by jpfr