open62541/open62541 #875: LDS-ME feature

Wtango, 2016-11-02:
Hi,
       I'm interested in LDS-ME feature, will open62541 support that feature recently? If yes, will it be a independent component or integrate into the SDK?

Thanks in advance.

jpfr, 2016-11-02:
@Pro did some great work in the #732 pull request.
It will be merged into master once we pushed 0.2 (and let the branches diverge more).

closed 2016-11-02 by jpfr

Pro, 2016-11-02:
@Wtango i try to keep the `feature_mdns` branch up to date with the 0.2 (https://github.com/open62541/open62541/tree/feature_mdns).
Feel free to try it and report issues here in the repo.

I tested it for a few weeks now and it is quite stable

Wtango, 2016-11-04:
@Pro , I try to build feature_mdns branch, but i get an error show that 
**CMake Error: File /home/tango/opcua/open62541/deps/mdnsd/libmdnsd/mdnsd_config.h.in does not exist.
CMake Error at CMakeLists.txt:178 (configure_file):
  configure_file Problem configuring file
**
Is “mdnsd_config.h.in” missing in the repository?

Pro, 2016-11-04:
Did you use?:
```
git clone https://github.com/open62541/open62541.git
git submodule update --init --recursive
```

libmdns is a submodule, thus you need to update the git submodule too.

It should build without any error:
https://travis-ci.org/open62541/open62541/builds/172638152

Wtango, 2016-11-04:
@Pro , i run the "discovery_server_discovery" then run the "discovery_server_register" program, it raise a segment fault.  trace:

**#0  0x00007fb6417fdcc0 in _IO_vfprintf_internal (s=0x7fb641b73620 <_IO_2_1_stdout_>, format=<optimized out>, ap=0x7ffd9b11fc78)
    at vfprintf.c:1632
#1  0x000000000041d1cd in UA_Log_Stdout (level=UA_LOGLEVEL_INFO, category=UA_LOGCATEGORY_USERLAND, msg=0x5dec38 "Node read %s")
    at /home/tango/opcua/open62541/plugins/ua_log_stdout.c:24
#2  0x00000000004018b6 in readInteger (handle=0x7ffd9b1201f0, nodeid=..., sourceTimeStamp=false, range=0x0, dataValue=0x7ffd9b11fdc0)
    at /home/tango/opcua/open62541/examples/discovery/server_register.c:44
#3  0x0000000000413a62 in UA_Server_addDataSourceVariableNode (server=0xe550a0, requestedNewNodeId=..., parentNodeId=..., 
    referenceTypeId=..., browseName=..., typeDefinition=..., attr=..., dataSource=..., outNewNodeId=0x0)
    at /home/tango/opcua/open62541/src/server/ua_services_nodemanagement.c:813
#4  0x0000000000401d27 in main (argc=1, argv=0x7ffd9b120638) at /home/tango/opcua/open62541/examples/discovery/server_register.c:89**

Pro, 2016-11-04:
thanks!
fixed in https://github.com/open62541/open62541/commit/d6c83743b9674d23285e82c6e80a856a3a597b1a