open62541/open62541 #879: cant connect Siemens Server

xty19, 2016-11-05:
I connect Siemens 840D opc ua server.
it can be connected via UA expert ,but cant via my code.
question is:
1.the program stuck at   **client->connection = client->config.connectionFunc(UA_ConnectionConfig_standard, endpointUrl, client->config.logger);**
2. could you please tell me how print the logger? i cant print it, thank you.

ps:this mac is not good to use ,i cant use markdown ,so pardon me

xty19, 2016-11-05:
[image]

xty19, 2016-11-07:
I can connect the siemens device ! :laughing: 
just change the UA_ClientCOnfig_standard !

`const UA_EXPORT UA_ClientConfig UA_ClientConfig_standard =
{
.timeout = 5000,
.secureChannelLifeTime = 600000,
.logger = Logger_Stdout,
.localConnectionConfig =
{
//original setting
/*.protocolVersion = 0,
.sendBufferSize = 65536,
.recvBufferSize = 65536,
.maxMessageSize = 65536,
.maxChunkCount = 1*/
- //changed

setting

.protocolVersion = 0,
.sendBufferSize = 16384,
.recvBufferSize = 16384,
.maxMessageSize = 65536,
.maxChunkCount = 0 },
.connectionFunc = UA_ClientConnectionTCP
};`

closed 2016-11-07 by xty19

Stasik0, 2016-11-07:
We need to handle it automatically. Could you give us a trace of errored connection?

xty19, 2016-11-07:
@Stasik0  it is my pleasure to give you them info,but i am a green hand,i dont konw how to get the info ,such as the logger and the shark software, would you please tell me how to use them and i can provide you the info. :smiley:

Stasik0, 2016-11-07:
Have a look in projects wiki

xty19, 2016-11-07:
@Stasik0  i am from China, the network cant visit wiki,google,youtube ect.  website :sob:

HolgerJeromin, 2016-11-07:
@xty19 Stasik did not wrote wikipedia but project wiki :-)
https://github.com/open62541/open62541/wiki/Capturing-traffic-with-Wireshark

xty19, 2016-11-07:
@HolgerJeromin thanks , i will read the page now, and do some work