open62541/open62541 #880: how to be a member of you?

xty19, 2016-11-08:
I am now using OPC-UA , and i have two device :FANUC, SIEMENS.
i am on duty for OPC-UA part in my team. all in all i use a lot, 
but i dont want to be a newer always, so i want to study deeper,
so how to be a member of you?
:smiley:

jpfr, 2016-11-11:
Hello @xty19 ,

like many open source projects, open62541 is meritocracy.
Giving commit access is decided based on past work.
That is developed features or being the de-facto maintainer of a subsystem.

However, you can contribute to open62541 right away. There is no need of an official membership.
Reporting issues is already a welcome contribution!
Code contributions can be sent as GitHub pull requests or by posting a diff.
The same holds for documentation improvements.

If you want to study OPC UA more closely, I suggest you download the standard from the OPC Foundation. It is also a good idea to compare implementations in terms of features / design.

jpfr, 2016-11-11:
I forgot to say: Welcome @xty19!
I'm looking forward to hear about your experience with open62541.

xty19, 2016-11-13:
@jpfr  
thanks for you reply
i Begin to be familiar with github fork, and wireshark, and i will contribute if i can :smiley: :

jpfr, 2016-11-17:
Great! Looking forward to seeing contributions from your side.

closed 2016-11-17 by jpfr