open62541/open62541 #881: Build problem on master (ua_config_standard.c:87:5: error: initializer element is not constant)

Schwane, 2016-11-08:
Hi,
I just pulled the current master branch from github. When I tried to build the project I got a compiler error.
gcc-version:

> gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.3)

compiler error:

> /home/sebastian/opcUa/open62541/plugins/ua_config_standard.c:87:5: error: initializer element is not constant
     },
     ^
/home/sebastian/opcUa/open62541/plugins/ua_config_standard.c:87:5: error: (near initialization for ‘UA_ServerConfig_standard.accessControl’)
make[2]: *** [CMakeFiles/open62541-object.dir/plugins/ua_config_standard.c.o] Error 1
make[1]: *** [CMakeFiles/open62541-object.dir/all] Error 2
make: *** [all] Error 2

I assume that it was introduced with commit 860b800b82b5f8b60fa4cd70611897eaa3b510c8
The last recent commit in my master branch is e5f5d03c8b326bd24a4a79d63b78d8b599223fa9.

Best regards,
Sebastian

jpfr, 2016-11-08:
Hey @Schwane,

thanks for reporting this. I'll see that the issue on gcc 4.8 is fixed.

It is however recommended that you use the 0.2 from which we will tag the next stable release.
We run continuous integration on 0.2 that includes building with gcc 4.8.

jpfr, 2016-11-21:
Fixed in 44a24c67d14630d0cf0503ba38a532641e2980bf

closed 2016-11-21 by jpfr

Schwane, 2016-11-21:
Hi,
I can confirm that.