open62541/open62541 #883: Is there any limit for the creating Object nodes at server

tijo40, 2016-11-10:
Hello,

I have a situation in which i am creating an object node with a variable node as child at the Server. When i tried loading 13200 object nodes, it works fine. But when I tried increasing the Object nodes to 18000, the application crashes with error details as "Not enough memory to complete the operation."

jpfr, 2016-11-14:
Hello @tijo40,

we try to fail gracefully when memory runs out. AFAIK, we check after every malloc if the returned value is NULL (no memory could be allocated).

Could the process have been killed by the operating system?
Please post error logs / and or a backtrace generated by a debugger.
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Otherwise, it is difficult for us to infer what is happening in your system.

Best regards,
Julius

tijo40, 2016-11-14:
Hello @jpfr ,

Sorry for not detailing the issue report earlier. 

	/* Basic Structure of Node Hierarchy
	 *
         * +- (OT) PluginType
	 *         |
         *        +- (OT) TableType     
	 *			 |
	 *		        + (V) Variable1
	 *  		        + (V) Variable2
         */ 

/* Instantiation 
	*
	* Folder1 <FolderType>
	* |	
	* +- Plugin1 <PluginType>
	* |    |
        * |    +- Table1 <TableType>     
	* |	|	|
	* |	|	+ (V) Variable1
	* |     |	+ (V) Variable2
	* |	|
	* |	+- Table2 <TableType>     
	* |		|
	* |		+ (V) Variable1
	* |  	        + (V) Variable2
	* |  
	* +- Plugin2 <PluginType>
	*      |
        *     +- Table1 <TableType>     
	*	|	|
	*	|	+ (V) Variable1
	*      |	+ (V) Variable2
	*	|
	*	+- Table2 <TableType>     
	*		|
	*		+ (V) Variable1
	*  	        + (V) Variable2

Before adding each child Object, I used to browse at the server side whether child node is present or not.  Previously, browse result & browse description memory was not deallocated at the end. So the memory allocation got increased and went to deadlock.

 Now I have added the delete members option, the memory is available now. I have tested and its working now.

Below mentioned is thefunction used to browse at the Server side.

/*!	
* \check node already created for a parent or not with NodeID as search criteria
*/
void
browseNodeIterator(UA_NodeId* theParentNodeID,
				   UA_NodeId* theChildNodeID,
				   bool& theResult)
{
    UA_BrowseDescription bDescription;
    UA_BrowseDescription_init(&bDescription);
    UA_NodeId_copy(theParentNodeID, &bDescription.nodeId);
    bDescription.referenceTypeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    bDescription.browseDirection = UA_BROWSEDIRECTION_FORWARD;
    bDescription.includeSubtypes = true;
    UA_BrowseResult browseResult;
    UA_BrowseResult_init(&browseResult);
    // UA_Server_browse(myOPCcommServer, 0, &bDescription);
    browseResult = UA_Server_browse(myOPCcommServer, UA_UINT32_MAX, &bDescription);
    UA_ReferenceDescription *referenceDesc = browseResult.references;
    bool tempResult = false;

    for(size_t iterator = 0;
        iterator < browseResult.referencesSize;
        ++iterator) {

        UA_NodeId tempID = referenceDesc->nodeId.nodeId;
        tempResult = UA_NodeId_equal(&tempID, theChildNodeID);

        if (tempResult){
            theResult = true;
            break;
        }
        ++referenceDesc;
    }

    if (!tempResult){
        theResult = false;
    }

    // Newly added to clear memory
    UA_BrowseResult_deleteMembers(&browseResult);
    UA_BrowseDescription_deleteMembers(&bDescription);
}

closed 2016-11-14 by tijo40