open62541/open62541 #884: Windows Embedded Compact 2013 build

carljoerger, 2016-11-10:
Hello,
   I've modified the openc62541.c and openc62541.h files to compile as C++ for Windows Embedded Compact 2013 (under VS2015).  Is anyone else interested in this?  I haven't tested them yet.  Please give me any feedback...

[open62541.zip](https://github.com/open62541/open62541/files/584286/open62541.zip)

jpfr, 2016-11-10:
Hey @carljoerger,

Great work!
Sure we are interested. 😄 

Can you tell us the version you started with so we can make a diff for preparing a commit?

If you want, you can also make a pull request on github, so your name appears in the repository as the author of the commit.

Stasik0, 2016-11-10:
The files are marked as Git-Revision: v0.2.0-RC1, so probably the last release.

carljoerger, 2016-11-10:
I want to port it to WEC 7 (VS 2008).
The designated initializers are the main problem (or just porting it to C++).  Is there any chance that they could be changed?

HolgerJeromin, 2016-11-10:
@jpfr with git you can change the author of a commit yourself. No need to have an own pr to clean credit a different author.

carljoerger, 2016-11-10:
How do you make a pull request?  I'd sure like them to be committed, but I hacked up the code pretty well - any feedback is appreciated.

jpfr, 2016-11-11:
open62541.c diff: http://paste2.org/9HBn82yj
open62541.h diff: http://paste2.org/MOXCtJHy

You changed quite a bit there. Can you summarize the changes?

I see you changed to open62541.cpp. Normally you can just compile with a C-compiler and link against the object file from your C++ project.

I'm unsure what to do with this. Merging is a major effort.
And we support Visual Studio 2015 anyway. (And the VS 2013 C compiler should be supported as well, imho). More arguments towards merging would be helpful. 😄 

Github is very well documented:
https://help.github.com/articles/creating-a-pull-request/

carljoerger, 2016-11-11:
My main problem was that the Windows Embedded Compact 2013 C compiler in VS2015 didn't seem to support the bool type!  So rather than define bool as char or something like that, I compiled the code using  the C++ compiler.  C++ doesn't support designated initializers, so I had to rearrange some of the initializer lists and remove the designated initializers.  Some of the structures were defined after they were used, so I had to move those definitions.  The C++ compiler complained about assigning void pointers to structure pointers, so I added the appropriate casts.
I'm going to recompile these two files with the regular Windows C compiler and see if the sample OPC server works with the created library.  And I'm going to try to compile them for Windows Embedded Compact 7.
I don't know how other users would feel about removing the designated initializers.  The other changes could be improvements in either case.

carljoerger, 2016-11-11:
The modified .cpp and .h files compile and seem to work with the sample server code under Win32.
I'm going to port the files back to Windows Embedded Compact 7, create a sample client and server, and try them on some real hardware (will take a while).  I'll keep you posted.  If someone could verify that the .cpp and .h files work correctly for more ambitious projects, that would be nice.  I know that other people would be interested in using OPC in WEC7 or WEC2013.  Maybe other people would be interested in compiling this under C++.

carljoerger, 2016-11-11:
[open62541.zip](https://github.com/open62541/open62541/files/586377/open62541.zip)
Here are the files that compile in C or C++ using the VS2015 Win32 compiler.  I'm not ready to port this to WEC7 (my boss wants me to wrap this up).  I'm going to test these using the sample client and server under WEC2013.

HolgerJeromin, 2016-11-11:
If the bool was the initial problem did you tried to ifdef and set to 0/1?

carljoerger, 2016-11-11:
I don't understand - I was going to try

#ifdef CE7
   #define bool char
#endif

but that didn't seem safe.  What are you suggesting instead?

Stasik0, 2016-11-11:
http://stackoverflow.com/a/18367061

suggests

typedef enum { false = ( 1 == 0 ), true = ( ! false ) } bool;

carljoerger, 2016-11-11:
Ok - that seems reasonable.  I'll try that (soon).

jpfr, 2016-11-12:
> I don't understand - I was going to try

> #ifdef CE7
#define bool char
#endif

> but that didn't seem safe. What are you suggesting instead?

On most platforms, bool is 8bit long (== char). Zero is false and everything else is true.
An enum is an int, so takes up more memory. Both char and enum booleans should work just fine.
That includes the || and && operators.

frax2222, 2016-11-16:
Hi Carl,
in a first step towards WEC7 I could compile the files you uploaded in Visual Studio Express 2008.
I had to set compiler option /TP and I had to replace the includes inttypes.h and stdbool.h with attached stdint.h.
I got some warnings but the linked exe did work, I could connect with UAExpert.
[stdint.zip](https://github.com/open62541/open62541/files/595562/stdint.zip)

frax2222, 2016-11-21:
Please find attached VS2008 project for Win and WEC7
[opcuaserver.zip](https://github.com/open62541/open62541/files/603652/opcuaserver.zip)

carljoerger, 2016-11-21:
Hi frax2222 - you used my files!  Thanks!  I won't have a CE machine to test them on until December 12th.  What happens next?  The code is still pretty hacked up from the last release.

martinkellers, 2016-11-25:
Hi 
I'm just hooking into this thread as well since we are also very interested in getting OPC UA functionality on Windows CE. We are currently running on a Windows CE 6.0 X86 platform and I have have downgraded the VS2008 project to a VS2005 project and recompiled your source against our sdk with succes and I'm now able to run the code on our target platform. I have also tried getting the client example to work and I'm able to run the client example in debug mode. However I keep getting a  BADCONNECTIONCLOSED 0x80ae0000 return value when trying to connect to a server from the client. I'm very new to OPC UA and I might be missing out on something very basic here.

We are moving to WEC7 in the near future so this port of Open62541 is very interesting for us as well.

Best regards
Martin

Stasik0, 2016-11-25:
hi @martinkellers, could you first test our client with your particular server from a "normal" windows or linux plattform to make sure its a Win CE issue.

jpfr, 2016-11-29:
Hello there,

I added basic support for VS2008 to the 0.2 branch.
- a workaround for the missing `stdint.h` that is used on older Visual Studio versions
- a cmake option `UA_COMPILE_AS_CXX` that compiles the library with the c++ compiler (advanced option, may need to be uncovered)

Now I get ~5 errors on most file on VS2008. Some are a bit more complicated.
Mostly errors are simple ones like `void*` casting and compound literals.

@carljoerger 
@martinkellers 
@frax2222 

Are you interested in preparing a pull request that brings your Windows Embedded patch into the main repository?
We can also do a "pilot" with just a few files. Then, our continuous integration infrastructure will tell us, whether the changes are compatible with all the other compilers in our automated tests.

carljoerger, 2016-11-29:
I would love to - I might not be able to get to this until the end of the year, though.

jpfr, 2016-11-30:
Sure. Take your time.

carljoerger, 2016-12-06:
I tried again to compile the library with the WEC7 C compiler, but not only is bool not defined, but const is not defined!  Do you have any ideas?
The code I submitted compiles in VS2008 Win32 with both the C and C++ compilers.  I'd love to put the whole thing into the main repository - is that ok?  What would I need to do?

uleon, 2016-12-06:
undefined bool: Do WEC7 / WEC13 know about _Bool (which is actually the C99 bool type)? Then a simple
```
#define bool _Bool
```
might help.

carljoerger, 2016-12-06:
Do you have any ideas for const?

uleon, 2016-12-06:
Well, you could try
```
#define const
```
As C99 uses const in a sense of read-only and not (as in C++) as a const expression that might be used for instance for instanziating an array, this should do the job just fine.

carljoerger, 2016-12-13:
The sample client and server code worked with this library on my x86 WEC7 processor.  Thank you, everyone!

carljoerger, 2016-12-13:
jpfr, what needs to be done to integrate this into the main code?

martinkellers, 2016-12-14:
@carljoerger nice work :-)
I would love to do a pre-test of your code on our Wince 6.0 target - just to see the samples working on our platform as well. 
Can you provide a zipped file containing your project - I will then try to compile your code against our sdk with vs2005 and do a test. 
I have a lot on my table right now - but hopefully I can try to test it out this week.

Best Regards
Martin

carljoerger, 2016-12-14:
[OPCCli.zip](https://github.com/open62541/open62541/files/651980/OPCCli.zip)
I created a library and linked the client and the server with it.  I used frax2222's project for ideas for some of the project settings.

carljoerger, 2016-12-14:
Now I'm trying to connect to my sample server over our network using the Opc.Ua.SampleClient.  My server name is opc.tcp://learycontroller:16664 and my network address is 192.161.1.157.  What am I missing for me to discover my server on the network?

carljoerger, 2016-12-14:
I was able to connect to it at opc.tcp://192.168.1.157:16664
Martin, I hope you get yours working as well.

jpfr, 2016-12-16:
Hello there,

I prepared a branch https://github.com/open62541/open62541/tree/master_vs2008 derived from master. It has Visual Studio 2008 enabled for continuous integration with appveyor (the oldest VS version available there). Of course, the integration tests on the branch will fail until the transition is complete.

So the scaffolding is ready and we can start working on getting this work into the master branch.
If you send pull requests, please consider the following:
- Make Pull Requests (PR) against the master_vs2008 branch. This is the only branch that has vs2008 testing enabled. We will merge into master from there.
- Only transition 2-3 files in every PR so that reviewing is possible
- We do not target Visual Studio (ANSI) C before VS2013. Only Visual Studio C++. So no need to move all variable definitions to the beginning of code blocks.
- In general, make the changes as non-invasive as possible. At best, start from a clean master_vs2008 version and **only fix the errors** so that 2-3 more files compile.

Thank you all. And keep up the good work!

Best regards,
Julius

frax2222, 2016-12-16:
Hello,

the main problem seem to be the struct initializers. VS2008 does not know .membername=value syntax even with option /TP

Any ideas ?

jpfr, 2016-12-16:
Jup. Designated initializers are not legal C++. Also not in C++11 and upwards.

Designated initializers don't **do** anything. They provide meta-information for us programmers.
So we don't get lost in complicated structures.

I'd say we have the following options. Which to use in which case is essentially a question of taste.

```c
// was
UA_String s = {.length = 1, .data = "a"};

// becomes
UA_String s;
s.data = "a";
s.length = 1;

// or
UA_String s = {
    1,  /* .length */
    "a" /* .data */
};

// and if there is really no possibility to mis-read the initialization
UA_String s = {1, "a"};
```

Stasik0, 2016-12-16:
+1 for

UA_String s = { 1, /* .length */
"a" /* .data */ };

Imho its less copy-paste error prone

> Am 16.12.2016 um 18:49 schrieb Julius Pfrommer <notifications@github.com>:
> 
> UA_String s = { 1, /* .length */ "a" /* .data */ };

carljoerger, 2016-12-16:
Is this legal for static data?

UA_String s;
s.length = 1;
s.data = "a";

jpfr, 2016-12-16:
> Is this legal for static data? And is it legal both in C and C++?
UA_String s;
s.length = 1;
s.data = "a";

No. This only works for structs that are created at runtime.

frax2222, 2016-12-18:
Hello,
most of these initializations are in ua-types-generated.c which is generated by generate_datatypes.py from opc.ua.types.bsd.
Is there a way to modify generate_datatypes.py so that it generates the initializations as VS2008 needs them ? (i.e. UA_String s = {1, "a"}; with initilizers in correct sequence)
I do not know Python so it is difficult for me to understand the code.

jpfr, 2016-12-22:
The two most relevant functions are `datatype_c` and ``members_c`.
https://github.com/open62541/open62541/blob/0.2/tools/generate_datatypes.py#L77

Both functions are simply constructing a string and return it.
The only uncommon syntax is the insertion of substrings:
```python
n = "name"
h = "Hello %s" % self.name 
```

If this is blocking your progress, please give us a short heads up.

carljoerger, 2016-12-22:
I haven't been doing any work on this - I hope that's not causing problems.

frax2222, 2016-12-23:
Hello jpfr,
you pointed to line 77 - this made me understand the code.
Thank you.
I am not going through files.....

[generate_datatypes.zip](https://github.com/open62541/open62541/files/670843/generate_datatypes.zip)

frax2222, 2016-12-23:
Hello,

I now nearly got it compiled with Visual Express 2008.

There are 2 remaining problems:

UA_Node * UA_NodeStore_newNode(UA_NodeClass class) {
ua_nodestore.c(208) : error C2332: 'class': missing tagname

and

queueItem = TAILQ_LAST(&mon->queue, QueueOfQueueDataValues);
ua_subscription.c(57) : error C2027: undefined type used "ensureSpaceInMonitoredItemQueue::QueueOfQueueDataValues"

I have uploaded the whole stuff and will make a Christmas break - maybe somebody has time to check and proceed...

[open62541-master_vs2008.zip](https://github.com/open62541/open62541/files/671417/open62541-master_vs2008.zip)

frax2222, 2016-12-27:
Hello,

plesae find attached  modified VS2008 branch.
use cmake_vs2008.bat to generate. amalgated files in build may be used for WEC7 (Compiler Option /TP)
[open62541-master_vs2008.zip](https://github.com/open62541/open62541/files/673803/open62541-master_vs2008.zip)

jpfr, 2017-01-01:
Thank you @frax2222.

We should merge this via a pull request.
If you haven't worked with github before, we can create one. If you want to see your name as author of the commit in the history, feel free the create a pull request yourself.

Anyway, let's see to get this included in the repository and the continuous integration builds!

frax2222, 2017-01-01:
Hello jpfr,

I haven't worked with github before so I think it is better this is done by someone who knows how to do this.
Thank you !

jpfr, 2017-01-08:
Hello there!

I pulled the patch in: https://github.com/open62541/open62541/tree/master_vs2008
Compiles on VS2008 without any warnings and runs fine. :-)

The example projects need some more work. There are linker errors remaining. But that is due to how CMake generates the projects, not the code.

The main library compiles fine. And I can manally set up a VS2008 project with it that works.
See also the attached file: [vs2008_testserver.zip](https://github.com/open62541/open62541/files/692093/vs2008_testserver.zip)

Therefore, in the near future **open62541 is no longer written in C99, but in the common subset of C99 and C++03**

frax2222, 2017-01-12:
Hello jpfr,

thank you for your work !
I have downloaded and generated amalgameted files for use with VS2008/WEC7.

I had do to some modifications:

in CMakelists.txt
-----------------
replace: if(NOT ${OPEN62541_VER_COMMIT})
with: if("${OPEN62541_VER_COMMIT}" STREQUAL "")

replace: option(UA_COMPILE_AS_CXX "Force compilation with a C++ compiler" OFF)
with: option(UA_COMPILE_AS_CXX "Force compilation with a C++ compiler" ON)

in deps\ms_stdint.h
--------------------
replace: #  include <wchar.h>
with:
#ifndef UNDER_CE
# include <wchar.h>
#endif

in src\ua_conncetion.c:
-----------------------
replace: *message = {length, data};
with:
UA_ByteString temp;
temp.length = length;
temp.data = data;
*message = temp;

Sorry - maybe I should learn how to work with github.....

HolgerJeromin, 2017-01-12:
You can edit source code right with the browser and create pull requests. Not as nice as with the git client or github windows software, but easy to use.

jpfr, 2017-01-13:
The branch passed all tests and is merged to master. Thanks for everybody involved!

@frax2222
Your patch is applied and on the master branch. However, I did not set C++ compilation as a default.
You will have to set that option with a command line argument or in cmake-gui.

closed 2017-01-13 by jpfr

frax2222, 2017-01-16:
hello,

there is no longer a vs_2008 branch so I downloaded master branch.
But I could not find the changes needed for VS2008 in master branch ?!

Pro, 2018-09-04:
Related to #2026

JesperVilladsen, 2019-03-18:
Hello,
I am trying to build Open62541 for winCE 6.0 in ws2005. Did anyone have success building this with version 0.3.0 of Open62541?
I am new to gitHub, so the solution might already existing without me knowing. Thank you