open62541/open62541 #885: Code 80740000 is appear

tpiat, 2016-11-14:
Previously the server was running correctly. After some changes stopped created arrays also does not work pasted from the example code. Error 80740000 when call UA_Server_addVariableNode. Can anyone suggest what the problem is?

tpiat, 2016-11-14:
In previous version works!
I don't know how to define version in zip file.

jpfr, 2016-11-14:
The error code you get is `UA_STATUSCODE_BADTYPEMISMATCH`.

There are now additionally consistency checks for variable nodes. Probably your dataType, valueRank or arrayDimensions attribute does not match with the value (or the variable type).

We can help more if you provide error logs and / or code snippets concerning the specific variable that cannot be added.
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

tpiat, 2016-11-14:
Thanx.
I wrote that pasted code do not works.
But the code works in the server.c demo sample.
Tomorrow I'll write logs.

closed 2016-11-15 by tpiat