open62541/open62541 #886: Nodes without parent in the nodeset2.xml

jpfr, 2016-11-16:
Hello all,

there are some nodes in the full nodeset2 xml without a hierarchical parent reference.
For example this one:

```xml
  <UAObject NodeId="i=78" BrowseName="Mandatory" SymbolicName="ModellingRule_Mandatory">
    <DisplayName>Mandatory</DisplayName>
    <Description>Specifies that an instance with the attributes and references of the instance declaration must appear when a type is instantiated.</Description>
    <References>
      <Reference ReferenceType="HasProperty">i=112</Reference>
      <Reference ReferenceType="HasTypeDefinition">i=77</Reference>
    </References>
  </UAObject>
```

How do we deal with that in the nodeset compiler? @ichrispa 
I have problems with the new consistency checks. Because that node should not be legal.
Maybe we have to give that one to the OPC Foundation issue tracker.

Best regards,
Julius

jpfr, 2016-11-16:
By the way, this is a blocker for #882, since it would break nodeset generation for the full nodeset2.

jpfr, 2016-11-16:
Hrmpf. It looks like there can be objects without a parent.
That's not cool. That ruins my assumptions on consistency. :smile:

closed 2016-11-16 by jpfr

Stasik0, 2016-11-16:
What about the root node?