open62541/open62541 #888: UA_InstantiationCallback issue

Wtango, 2016-11-20:
Hi,

On 0.2 branch,I create a instance of FileType(0,11575) and set the "UA_InstantiationCallback".
In the callback method, I print out all nodeId get from parameters what shows that:

> newNodeId:1860		templateId:68
newNodeId:1860		templateId:11576
newNodeId:1861		templateId:68
newNodeId:1861		templateId:11577
newNodeId:1862		templateId:68
newNodeId:1862		templateId:11578
newNodeId:1863		templateId:68
newNodeId:1863		templateId:11579
newNodeId:1859		templateId:11575

I can not get the any MethodNode(eg. open() ) from the callback function.
And the new object i created, it's MethodNode reference the same node as FileType's.

jpfr, 2016-11-21:
Methods are not instantiated.
It's okay to just link new objects to the existing method.
The method will get the object "context" as the first argument.

Wtango, 2016-11-21:
But i see in that in unified automation C++ SDK, will create a new MethodNode for each instance object.
And what is it use for calling the InstantiationCallback with parameters "newNodeId:1861	templateId:68"?

jpfr, 2016-11-21:
The standard (part 3) says:

> A Method shall always be the TargetNode of at least one HasComponent Reference. The
SourceNode of these HasComponent References shall be an Object or an ObjectType. If a
Method is called then the NodeId of one of those Nodes shall be put into the Call Service defined
in Part 4 as parameter to detect the context of the Method operation.

So there can be several objects with the same method as component. Creating a copy of the method seems to be valid as well.

I'll have a look at the instantiation callback.

closed 2021-11-23 by jpfr