open62541/open62541 #889: Subscription issue when request over 600 items.

JinGuoGe, 2016-11-21:
1) subscription issue
I requested monitoring request over 600 items. But  get timeout response.
But success when request less 600 items. 
I am using only open62541.h/open62541.c source file for creating opc server and opc client.

2) multithread issue
When I enable multithreading  in open62541.h,  even mornitoring  request for one item doesn't work.
     #define UA_ENABLE_MULTITHREADING

jpfr, 2016-11-24:
Hey @JinGuoGe ,

can you try with a recent nightly build from here?
http://open62541.org/releases/
Handling of repeated jobs improved quite a bit recently.

I will look into the multithreading issue.

If the issue persists, can you provide debug logging output so we can pinpoint the issue cause?
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Best regards,
Julius

JinGuoGe, 2016-11-29:
Hi, Julius.
Thanks for your solution.
When I used updated version from releases, it works well. 
Really thanks. 
And If multithreading issue is solved, it will be great help.

Best regards.
GuoGe Jin

closed 2016-11-30 by jpfr

TianDaGe, 2016-12-02:
Hi did you solve multithreading issue?

jpfr, 2016-12-02:
I fixed an issue and multithreading now works for me.

However, we have no enough tests to declare multithreading "bug-free".
I just marked it as an experimental feature.