open62541/open62541 #892: UA_Server_new() gives 'General Protection' fault under CLANG 3.3

escherstair, 2016-11-28:
Trying to run [server_firstSteps](https://github.com/open62541/open62541/blob/0.2/examples/server_firstSteps.c) example, compiled with CLANG 3.3, UA_Server_new() gives 'General Protection' fault because (I think) the following instruction is not supported
`namespaceArray->value.dataSource = (UA_DataSource) {.handle = server, .read = readNamespaces,
                                                        .write = NULL};`
These are the following lines [925](https://github.com/open62541/open62541/blob/0.2/src/server/ua_server.c#L925) and [926](https://github.com/open62541/open62541/blob/0.2/src/server/ua_server.c#L926).

Do you think it's possible to replace this kind of struct-members setup with a clang-compliant way?

escherstair, 2016-11-28:
After a deeper investigation, this is related to designated initializers that **are** supported by CLANG 3.3.
I'm going to dig into my specific CLANG compiler

Stasik0, 2016-11-28:
Otherwise maybe alloca and normal inits?

escherstair, 2016-11-29:
It could be an idea, but I prefer not to modify open62541 sources to fix an issue in my compiler.
Give me some time to investigate

jpfr, 2017-01-06:
Hello @escherstair ,

could you dig into this a little more?
The code section you indicate is perfectly legal C99. If clang parses that, it should not generate invalid code.

You could
- run with gdb
- run with valgrind
to get a better report of the error.

escherstair, 2017-01-09:
Since the code is perfectly legal C99 I think the problem is in the porting of CLANG used by my IDE.
If the rules of this forum allow me to write the name of this IDE I can share this info (it's a commercial product).
I've already forwarded this info to the IDe producer and they're going to work over it.

jpfr, 2017-01-09:
There's nothing that prevents you naming software on Github,
We understand if you choose not to disclose information pertaining to your business.

Anyway, please let us know if something interesting comes out of this investigation.
We can improve that piece of code with some other legal C99.
But we are not in a position to verify what would work for your Clang derivative.

escherstair, 2017-01-09:
Ok. I use LabWindows/CVI 2015 SP1 from National Instruments. It compiles the sources using a compiler based on CLANG 3.3.
I've already verified that 'Designated Inits' should be supported by the NI porting of CLANG 3.3, but the produce the issue. If i change to code to alloca() + normal inits the issue moves to the next usage of 'Designated Inits'.
You can find [here](http://forums.ni.com/t5/LabWindows-CVI/Designated-Inits-in-CVI-2015-SP1/m-p/3547759) the discussion I started on NI forum (my nickname is different on that forum).
As far as I know this library is checked with clang using travis system and unit testing approach and so I think NI should fix the issue in their porting (since I agree the syntax is legal C99).

jpfr, 2017-01-09:
Hello @escherstair ,

so this is one of the very rare cases where it's actual the compiler's fault. 😄 

Designated initialisers are on the way out in open62541.
The users @carljoerger and @frax2222 contributed a port of open62541 to a subset of both C99 and C++ to support older compilers. Since C++ does not support designated initializers (they prefer constructor methods, also for structs), we will go back to ANSI C initializers only. The port of the master branch is already done and my local tests all run fine. There are just some last adjustments to be made to our continuous integration infrastructure to test the C++ build as well.

So feel free to use https://github.com/open62541/open62541/tree/master_vs2008 if you continue to have problems with your Clang derivative. I can also give a short hint once the port makes it into master.

escherstair, 2017-01-12:
Is there an easy and quick way to get single the single pair of header and source files of the library from branch [https://github.com/open62541/open62541/tree/master_vs2008](url)?

jpfr, 2017-01-12:
We create nightly builds for win32.
http://open62541.org/releases/

Take the top entry from master_vs2008. The single-file sources are in the zip.

jpfr, 2017-01-12:
Hmm. It seems that posting nightly builds broke two weeks ago.
I keep you posted.

jpfr, 2017-01-13:
Nightly single-file releases are back: http://open62541.org/releases/

jpfr, 2017-01-13:
Closing the issue, as the master branch uses no designated initializers any more.
Please reopen if necessary.

closed 2017-01-13 by jpfr