open62541/open62541 #897: Issue when expanding namespace index

JinGuoGe, 2016-12-02:
I'm using open62541.c/open62541.h for building my example opcserver.
I'm going to use namespaceindex =2 (or 3,4...).

 /* 2) Add the variable node to the information model */
    UA_NodeId myIntegerNodeId = UA_NODEID_STRING(2, "the.answer");
    UA_QualifiedName myIntegerName = UA_QUALIFIEDNAME(2, "the answer");

When I use 1 instead of 2, then there is no problem, but when using 2,  node is not created.
So I analyzed open62541.c source, I found following code.

server->namespacesSize = 2 // Here I changed from 2 to 3.
After than created node, but when stopping server by ctrl+c key, generate following log.
   Segmentation fault (core dumped)
What is the problem? It is because of changing 2 to 3. Then cannot expand namespace index anymore?

Looking forward your answer.
Best regards.

JGrothoff, 2016-12-02:
Hi, you have to add a new namespace after the instanciation of the server(UA_Server_new(...)) and before inserting nodes to it (UA_Server_addVariableNode(...)) with:

UA_UInt16 newNamespaceIndex = **UA_Server_addNamespace**(server, "http://nameOfNamespace"); 

Explanation:
When you create a UA_Server with UA_Server_new(...) it has two namespaces per default. (Namespace 0 for the basic informationmodel and namespace 1 as default user namespace.) Thats where the line "server->namespacesSize = 2" comes from. It is the initialization of the servers namespace array.

JinGuoGe, 2016-12-02:
Thanks for your answer. Solved issue. thanks.

closed 2016-12-02 by JinGuoGe