open62541/open62541 #898: Multithreading compilation issue in Ubuntu16.04

tijo40, 2016-12-02:
Hello,

I am building with qt creator  in Ubuntu 16.04. When multithreading enabled, getting undefined reference error for liburcu.  

/usr/include/urcu/static/urcu.h:97: error: undefined reference to `rcu_has_sys_membarrier_memb'
/usr/include/urcu/static/urcu.h:198: error: undefined reference to `rcu_reader_memb'
/usr/include/urcu/static/urcu.h:198: error: undefined reference to `rcu_gp_memb'
/usr/include/urcu/static/urcu.h:201: error: undefined reference to `rcu_reader_memb'
/usr/include/urcu/static/urcu.h:220: error: undefined reference to `rcu_reader_memb'

/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:15640: error: undefined reference to `rcu_init_memb'
/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:17454: error: undefined reference to `rcu_register_thread_memb'
/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:17476: error: undefined reference to `rcu_barrier_memb'
/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:17477: error: undefined reference to `rcu_unregister_thread_memb'
/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:18004: error: undefined reference to `rcu_barrier_memb'

/usr/include/urcu/rculfhash.h:166: error: undefined reference to `rcu_flavor_memb'
/usr/include/urcu/rculfhash.h:166: error: undefined reference to `_cds_lfht_new'

/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:18896: error: undefined reference to `cds_lfht_first'
/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:18898: error: undefined reference to `cds_lfht_del'
/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:18901: error: undefined reference to `call_rcu_memb'
/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:18903: error: undefined reference to `cds_lfht_next'
/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:18905: error: undefined reference to `cds_lfht_destroy'
/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:18933: error: undefined reference to `cds_lfht_add_unique'
/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:18947: error: undefined reference to `cds_lfht_count_nodes'
/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:18953: error: undefined reference to `cds_lfht_add_unique'
/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:18970: error: undefined reference to `cds_lfht_lookup'
/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:18980: error: undefined reference to `cds_lfht_replace'
/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:18987: error: undefined reference to `call_rcu_memb'
/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:18996: error: undefined reference to `cds_lfht_lookup'
/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:18997: error: undefined reference to `cds_lfht_del'
/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:19000: error: undefined reference to `call_rcu_memb'
/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:19009: error: undefined reference to `cds_lfht_lookup'
/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:19021: error: undefined reference to `cds_lfht_lookup'
/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:19040: error: undefined reference to `cds_lfht_first'
/home/tthomas/Documents/QtProjects/SVN/linux/M5_HMI/Includes/open62541/open62541.c:19044: error: undefined reference to `cds_lfht_next'

:-1: error: collect2: error: ld returned 1 exit status

Below mentioned is the installed Versions of liburcu

tthomas@tthomas-VirtualBox:~$ apt-show-versions liburcu-dev
liburcu-dev:amd64/xenial 0.9.1-3 uptodate
liburcu-dev:i386 not installed

tthomas@tthomas-VirtualBox:~$ apt-show-versions liburcu4
liburcu4:amd64/xenial 0.9.1-3 uptodate
liburcu4:i386 not installed

tthomas@tthomas-VirtualBox:~$ apt-show-versions liburcu4-dbgsym
liburcu4-dbgsym:amd64 0.9.1-3 installed: No available version in archive

GCC Version 

tthomas@tthomas-VirtualBox:~$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/5/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 5.4.0-6ubuntu1~16.04.4' --with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-5 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4) 

qt4 Qmake is used for generating make 

If multithreading not enabled, the compilation works fine. But i want to enable multithreading in the Project.

jpfr, 2017-02-08:
Can you show the list of libraries you give gcc in the command line argument?
We use the following to compile the examples: `urcu-cds urcu urcu-common`

Pro, 2017-02-20:
I'm using ubuntu 16.04 and it's working without any issue.

Using CMake it generate for example the following cc call:
```
/usr/lib/ccache/cc  -g   CMakeFiles/server.dir/server.c.o ../CMakeFiles/open62541-object.dir/src/ua_types.c.o ../CMakeFiles/open62541-object.dir/src/ua_types_encoding_binary.c.o ../CMakeFiles/open62541-object.dir/src_generated/ua_types_generated.c.o ../CMakeFiles/open62541-object.dir/src_generated/ua_transport_generated.c.o ../CMakeFiles/open62541-object.dir/src/ua_connection.c.o ../CMakeFiles/open62541-object.dir/src/ua_securechannel.c.o ../CMakeFiles/open62541-object.dir/src/ua_session.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_server.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_server_binary.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_server_utils.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_server_worker.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_securechannel_manager.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_session_manager.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_nodes.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_nodestore.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_nodestore_concurrent.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_services_discovery.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_services_securechannel.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_services_session.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_services_attribute.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_services_nodemanagement.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_services_view.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_services_call.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_subscription.c.o ../CMakeFiles/open62541-object.dir/src/server/ua_services_subscription.c.o ../CMakeFiles/open62541-object.dir/src/client/ua_client.c.o ../CMakeFiles/open62541-object.dir/src/client/ua_client_highlevel.c.o ../CMakeFiles/open62541-object.dir/src/client/ua_client_highlevel_subscriptions.c.o ../CMakeFiles/open62541-object.dir/plugins/ua_network_tcp.c.o ../CMakeFiles/open62541-object.dir/plugins/ua_accesscontrol_default.c.o ../CMakeFiles/open62541-object.dir/plugins/ua_clock.c.o ../CMakeFiles/open62541-object.dir/plugins/ua_log_stdout.c.o ../CMakeFiles/open62541-object.dir/plugins/ua_config_standard.c.o ../CMakeFiles/open62541-object.dir/deps/libc_time.c.o ../CMakeFiles/open62541-object.dir/deps/pcg_basic.c.o ../CMakeFiles/open62541-object.dir/src_generated/ua_statuscode_descriptions.c.o  -o ../bin/examples/server  -lpthread -lrt -lurcu-cds -lurcu -lurcu-common
```

Hope this helps. If not, please reopen the issue.

closed 2017-02-20 by Pro