open62541/open62541 #901: error/client	Reply answers the wrong request. Expected ns=0,i=676. But retrieved ns=0,i=634

symgyyb, 2016-12-06:
hi:
    when I was in call UA_Client_writeValueAttribute,If I call continuously ,sometimes is an error,sometimes is normal.The error information is as follows:
    [12/06/2016 07:57:37.420] error/client	Reply answers the wrong request. Expected ns=0,i=676. But retrieved ns=0,i=634
[12/06/2016 07:57:37.423] info/channel	Connection 17 | SecureChannel 8 | The sequence number was not increased by one. Got 48460, expected 48459
[12/06/2016 07:57:37.423] info/client	Error receiving the response

  hope for your help.thanks.

jpfr, 2016-12-06:
Hey @symgyyb ,

and welcome to this Github board.

Can you give as some more information?
Which version of open62541 are you using? Which configuration?

https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

jpfr, 2017-01-17:
Closing the issue for now.
We cannot procede due to lack of details for reproducing the issue.

closed 2017-01-17 by jpfr