open62541/open62541 #902: [question] No variant equality test function - shall I submit one?

ben-farnham, 2016-12-06:
Hello,

**Background**: I contribute to the [quasar project](https://github.com/quasar-team/quasar), an open source framework for rapid OPC-UA server development. Quasar supports using open62541.org as one of its OPC-UA 'backend' options; so open62541 provides OPC-UA access, mediated via a quasar compatibility layer ([open62541-compat](https://github.com/quasar-team/open62541-compat)).

The quasar compatilibilty layer C++ API requires an implementation for UaVariant equality (i.e. an operator== implementation). I've added equality check code to the quasar compatibility module, however, I wonder whether this equality check would be better added to open62541 itself. Would this benefit other open62541 users?

**Question**: is there interest in having a method 
`static UA_INLINE UA_Boolean UA_Variant_equal(const UA_Variant *v1, const UA_Variant *v2)`
that returns true if the variants pointed to by v1 and v2 are equivalent? If so please let me know, I'd be happy to contribute.

Great project by the way.
Ben

jpfr, 2016-12-06:
Hello Ben,

that sounds fantastic!

We have equality testing for some builtin types, e.g. Strings and NodeIds.
I guess we would need such a testing method for all the builtin types.

Then we only need a shortish function that operates on the generic type description we generate for all types. See e.g. here: https://github.com/open62541/open62541/blob/0.2/src/ua_types.c#L836

Is it okay if you make an eventual PR for the master branch?
0.2 is officially frozen and we don't want to introduce more blockers for a release.

ben-farnham, 2016-12-07:
Hi,

OK, great - thanks for the link, I'll take a look at that and the existing equality tests before I start.  I'll make a pull request on master when it's ready.

Ben

closed 2016-12-07 by ben-farnham