open62541/open62541 #903: Asynchronous server methods

paroga, 2016-12-06:
Compared to the official stack ([opcua_serverapi.c](https://github.com/OPCFoundation/UA-AnsiC/blob/9e5d1b319e14f7c21f04ac16c42a8c1cd18fe6ac/Stack/proxystub/serverstub/opcua_serverapi.c)) open62541 ([ua_server_binary.c](https://github.com/open62541/open62541/blob/542c43fe7cbc8e407a916c4010fc29baa6b1460c/src/server/ua_server_binary.c#L481)) does not provide any asynchronous methods at the server.
Are there any plans to implement that and if not, would you accept a PR to support that?
A first step would be to move the `UA_SecureChannel_sendBinaryMessage`in a new "finish service"-function (which will be called at the end of the various `UA_Service`-functions) and make sure that all pointers stay valid for the response.

jpfr, 2016-12-06:
Hey @paroga,

we currently have one service that does async responses: the publish service for subscriptions.

The change is pretty simple. Just send the response from within the service and not at the end of `processMSG`. Then we can delay sending the response as we please.

The nontrivial part is creating a separate thread that runs the method. And that can be killed via the *Cancel* service. If we don't have that, there will be no advantage over syncrhonous processing.

Note that we have a multithreading extension of open62541. There, we can already process requests in parallel. And that works already for methods, since the received messages are dispatched to worker queues that run `processMSG` each. And send responses each.

Can you explain your use case? Then we know better how to proceed.

paroga, 2016-12-06:
I'd like to do asynchronous processing in a single-threaded environment. In a first step I'd like to support a "two thread"-environment: (1) the open62541-thread handling the network stuff and (2) a "worker"-thread, which does the business-logic.
E.g. for a ReadRequest I'd like to dispatch the request from open62541 to the embedding application. When the application processed the request in it's own event queue it will return the result to open62541, which will send it to the client. It's mainly interesting for supporting `UA_ExternalNodeStore` with additional (or changed) methods, which allow asynchronous processing.

mustafacamurli, 2018-12-26:
Hi all,

I haven't seen any progress regarding this change request.
If nobody is working on it, I can integrate necessary changes as since @jpfr have mentioned that's pretty simple.

AugustinusVonHippo, 2019-05-16:
+1 I would appreciate a pull request for this feature, too.

aminosu, 2019-07-05:
Hi all,
we also would need this feature,
is there some Work In Progress or any plan regarding this?
Thank you in advance.

jpfr, 2019-09-03:
We have an implementation for this.
Will land on master in the next weeks.

jpfr, 2019-11-21:
This is available now.

closed 2019-11-21 by jpfr