open62541/open62541 #905: Event support in the client

jpfr, 2016-12-06:
As Requested here:
https://groups.google.com/forum/#!topic/open62541/sHGkoEMb-1M

See also the description of the solution approach in the mailing list disscussion.

This is up for grabs. Feel free to do something about a feature you are missing. 😄

fmeerkoetter, 2017-08-15:
I am currently trying to pick up from where the google groups discussion stopped.

Stasik0, 2017-08-15:
@jpfr what would be a nice roadmap for events? Maybe to get away from internal polling for subscriptions to an alternative triggerUpdate mechanism?

fmeerkoetter, 2017-08-15:
My current status is that i have something that is somewhat working. My next step would be to clean it up and to create a WIP-MR so we can have a discussion about the patchset.

Pro, 2017-08-17:
PR #1161

Pro, 2017-08-24:
Added with #1161

closed 2017-08-24 by Pro