open62541/open62541 #908: [Question] about sending binary data between opcserver and opcclient.

JinGuoGe, 2016-12-07:
Hi,
For example , I am going to send binary file content from opcclient to opcserver.
But there is not node type for binary data, so I can not create node for binary data.
How can I implement this other than encrypting file contents by base64?
Looking forward your solution.
Thanks.

jpfr, 2016-12-07:
Just create a variable node that contains a ByteString value.
ByteStrings can hold any string and also null entries. Base64 is not needed.

If your file is really big. Say, hundreds of megabyte, there is a mechanism that exposes filesystem-like operations (open / seek / read / write / close / ...). But I guess you won't need that.

JinGuoGe, 2016-12-07:
thanks for your solution.

closed 2016-12-07 by JinGuoGe