open62541/open62541 #910: Segmentation fault: when exceeds LifeTimeCount during processing subscription in opc server

JinGuoGe, 2016-12-09:
I am using 2cabd7c5d9(2016-12-07 11:18:56 +0100)  from http://open62541.org/releases.
I built simple opcserver and opcclient using single open62541.h/open62541.c.

//opcserver.c code part
void *server_startup_thread(void *ptr) {
	// set opcserver configuration to g_server.
	// Create and initialize node information for g_server
	...
    UA_StatusCode ret_run = UA_Server_run(g_server, &g_is_server_running);

    return (void *)ret_run;
}

int main()
{

 pthread_create( &g_server_thread, NULL, server_startup_thread, NULL);
 
 while (running == true) {
		for (i=0;i<CODE_COUNT;i++)
		{
			float f_value;
			f_value = rand() & 10000 + 1;
                         UA_Variant node_value;
                         char str_code[] = "xxx"; // get str_code for every i.
                         UA_NodeId data_node_id = UA_NODEID_STRING(OPC_NAMESPACE_INDEX, str_code);
			UA_Variant_setScalar(&node_value, &f_value, &UA_TYPES[UA_TYPES_FLOAT]);

			UA_Server_writeValue(g_server, data_node_id, node_value);
		}
		sleep(1);
    }
}
Then in opcclient side, send monitor request to opcserver.
Then it works well, get updated data from opcserver.

But problem  is from now.
If I disable network of computer which is running opcserver, then after some time, I get following error.
Please see http://paste2.org/HnhpBK8M
I also test with other open sources for opcclient and got same result. (https://github.com/eclipse/milo/tree/master/milo-examples/client-examples/src/main/java/org/eclipse/milo/examples/client)

Looking forward your answer.
Thanks.

Stasik0, 2016-12-09:
please run the server in valgrind or gdb to get the stack trace

jpfr, 2016-12-17:
Hey @JinGuoGe ,

I cannot reproduce. I disconnected the client multiple times and let the session time out.
No problems were indicated.

Can you please
- tell us the version you are using
- retry with 0.2-rc2 or the current 0.2 branch
- provide valgrind logs?

Otherwise, it will be very hard, if not impossible, to find the source of your issue.

Thanks and best regards,
Julius

JinGuoGe, 2017-01-06:
Hi, thanks for your analyzing.
I tested with  current 0.2 branch,  I also cannot reproduce issue.
So i think this issue was already solved.

Best regards.
Jin

closed 2017-01-06 by JinGuoGe