open62541/open62541 #911: [Raspberry PI][Raspbian][0.2-rc2]: Tests failed

miraenator, 2016-12-12:
Hi, 

when compiling under debian jessie (tried the same conf, v0.2-rc2), I get different results when running `make test` for x86_64 and arm6 architectures. Under x86_86 (Docker; `make && make test`) everything is OK), on Raspberry Pi it fails (17 failed tests of 18).

I've tried the same configuration/build commands:
***
`cmake -D UA_BUILD_UNIT_TESTS=ON  -D UA_ENABLE_AMALGAMATION=ON ..`

`make && make test`

***
x86_64 results: All OK (omitting details as unimportant)
***
Raspberry PI (only make test results):
...
```[100%] Built target check_utils
Running tests...
Test project /home/pi/open62541-0.2-rc2/build
      Start  1: types_builtin
 1/18 Test  #1: types_builtin .........................***Exception: Illegal  8.47 sec
      Start  2: types_memory
 2/18 Test  #2: types_memory ..........................***Exception: Illegal  8.22 sec
      Start  3: types_range
 3/18 Test  #3: types_range ...........................***Failed   13.35 sec
      Start  4: chunking
 4/18 Test  #4: chunking ..............................***Exception: Illegal  8.37 sec
      Start  5: services_view
 5/18 Test  #5: services_view .........................   Passed    8.49 sec
      Start  6: services_attributes
 6/18 Test  #6: services_attributes ...................***Exception: Illegal  8.63 sec
      Start  7: services_nodemanagement
 7/18 Test  #7: services_nodemanagement ...............***Exception: Illegal  8.47 sec
      Start  8: check_services_subscriptions
 8/18 Test  #8: check_services_subscriptions ..........***Exception: Illegal  8.51 sec
      Start  9: nodestore
 9/18 Test  #9: nodestore .............................***Exception: Illegal  8.19 sec
      Start 10: session
10/18 Test #10: session ...............................***Failed   11.10 sec
      Start 11: check_server_jobs
11/18 Test #11: check_server_jobs .....................***Exception: Illegal  8.49 sec
      Start 12: check_server_userspace
12/18 Test #12: check_server_userspace ................***Failed    9.72 sec
      Start 13: check_server_binary_messages_browse
13/18 Test #13: check_server_binary_messages_browse ...***Exception: Illegal  8.28 sec
      Start 14: check_server_binary_messages_read
14/18 Test #14: check_server_binary_messages_read .....***Exception: Illegal  8.33 sec
      Start 15: check_server_binary_messages_write
15/18 Test #15: check_server_binary_messages_write ....***Exception: Illegal  8.29 sec
      Start 16: check_client
16/18 Test #16: check_client ..........................***Exception: Illegal  8.43 sec
      Start 17: check_client_subscriptions
17/18 Test #17: check_client_subscriptions ............***Exception: Illegal  8.41 sec
      Start 18: check_utils
18/18 Test #18: check_utils ...........................***Exception: Illegal  8.20 sec

6% tests passed, 17 tests failed out of 18

Total Test time (real) = 160.26 sec

The following tests FAILED:
          1 - types_builtin (ILLEGAL)
          2 - types_memory (ILLEGAL)
          3 - types_range (Failed)
          4 - chunking (ILLEGAL)
          6 - services_attributes (ILLEGAL)
          7 - services_nodemanagement (ILLEGAL)
          8 - check_services_subscriptions (ILLEGAL)
          9 - nodestore (ILLEGAL)
         10 - session (Failed)
         11 - check_server_jobs (ILLEGAL)
         12 - check_server_userspace (Failed)
         13 - check_server_binary_messages_browse (ILLEGAL)
         14 - check_server_binary_messages_read (ILLEGAL)
         15 - check_server_binary_messages_write (ILLEGAL)
         16 - check_client (ILLEGAL)
         17 - check_client_subscriptions (ILLEGAL)
         18 - check_utils (ILLEGAL)
Errors while running CTest
Makefile:137: recipe for target 'test' failed
make: *** [test] Error 8
```
***
x86_64 (Docker Jessie) (make test: OK):
`Linux 17c1f20fa22d 3.16.0-4-amd64 #1 SMP Debian 3.16.36-1+deb8u2 (2016-10-19) x86_64 GNU/Linux`
***
Raspberry Pi (make test: Errors):
$ uname -a
`Linux raspberrypi 4.4.36+ #933 Fri Dec 2 21:57:25 GMT 2016 armv6l GNU/Linux`
$ cat /proc/cpuinfo 
```
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2708
Revision        : 000e
```
***
Adding Dockerfile that works (x86_64). I've replicated the same commands/libraries_install on RPi.
```
FROM debian:jessie
RUN apt-get clean && apt-get update && apt-get install -y  --no-install-recommends \
	build-essential gcc cmake \
	liburcu-dev \
	check pkg-config libsubunit-dev python valgrind \	
	wget ca-certificates

ENV RELEASE 0.2-rc2

RUN wget "https://github.com/open62541/open62541/archive/${RELEASE}.tar.gz" -O - | \
	tar zxC /
WORKDIR /open62541-${RELEASE}
#Needed to fix character read in amalgation: was ascii coded, we need utf-8
RUN sed -i 's/import io/import io\nimport codecs/' tools/amalgamate.py && \
	sed -i 's#= io.open#= codecs.open#' tools/amalgamate.py && \
	sed -i "s/'w'/'w', 'utf-8'/" tools/amalgamate.py

RUN mkdir build && cd build && \
	cmake \
		-D UA_BUILD_UNIT_TESTS=ON \
		-D UA_BUILD_EXAMPLES=ON \
		-D UA_ENABLE_AMALGAMATION=ON \
	..

RUN cd build && make
RUN cd build && make test
```
***
Any hints? Or suggestions to provide more detailed info?

**Note**: I have turned off the multithreading. When UA_ENABLE_MULTITHREADING=ON, the tests fail on both architectures (!).

jpfr, 2016-12-15:
What is the result when you manually run one of the tests?
`./tests/check_types_builtin`

It's possible we need to put valgrind-based testing behind a cmake option.

jpfr, 2016-12-15:
The current 0.2 branch has moved valgrinded unit tests behind an option that needs to be activated.
Can you try again with / without valgrind?

Thanks @miraenator

miraenator, 2016-12-15:
Hi, 

it resolved (hid) the issue. (Valgrind still fails; posting details as a new post).

```
git pull (commit 6577996af526cb0d6b1091204dd7c28b76356a7c)
make clean
cmake -D UA_ENABLE_AMALGAMATION=ON -DUA_BUILD_UNIT_TESTS=ON ..

make && make test
```
works.

```
100% tests passed, 0 tests failed out of 18

Total Test time (real) =  13.78 sec
```

---
on
```
cmake -D UA_ENABLE_AMALGAMATION=ON -DUA_BUILD_UNIT_TESTS=ON -DUA_ENABLE_VALGRIND_UNIT_TESTS=ON ..
make && make test
```
it fails (as expected):
```
6% tests passed, 17 tests failed out of 18
```
---
However, if I run them separately, it is OK:
```
$ for i in tests/check_* ; do echo -e "\n\n $i" ; ${i} && echo "${i}: ====OK===" ; done

 tests/check_chunking
Running suite(s): Chunked encoding
100%: Checks: 2, Failures: 0, Errors: 0
tests/check_chunking: ====OK===

 tests/check_client
Running suite(s): Client
[12/15/2016 15:08:38.860] info/network  TCP network layer listening on opc.tcp://raspberrypi:16664
[12/15/2016 15:08:38.871] info/network  Connection 6 | New connection over TCP from 127.0.0.1:40484
[12/15/2016 15:08:38.876] info/channel  Connection 6 | SecureChannel 1 | OpenSecureChannel: Opened SecureChannel
[12/15/2016 15:08:38.884] info/session  Connection 6 | SecureChannel 1 | Session {1CD777E3-9590-A4E2-6DB6-4BE40EE09EB9} | ActivateSession: Session activated
[12/15/2016 15:08:38.889] info/session  Connection 6 | SecureChannel 1 | Session {1CD777E3-9590-A4E2-6DB6-4BE40EE09EB9} | CloseSession
[12/15/2016 15:08:38.893] info/channel  Connection 6 | SecureChannel 1 | CloseSecureChannel
[12/15/2016 15:08:38.898] info/network  Shutting down the TCP network layer with 1 open connection(s)
100%: Checks: 1, Failures: 0, Errors: 0
tests/check_client: ====OK===

 tests/check_client_subscriptions
Running suite(s): Client Subscription
[12/15/2016 15:08:38.943] info/network  TCP network layer listening on opc.tcp://raspberrypi:16664
[12/15/2016 15:08:38.958] info/network  Connection 6 | New connection over TCP from 127.0.0.1:40486
[12/15/2016 15:08:38.964] info/channel  Connection 6 | SecureChannel 1 | OpenSecureChannel: Opened SecureChannel
[12/15/2016 15:08:38.971] info/session  Connection 6 | SecureChannel 1 | Session {1CD777E3-9590-A4E2-6DB6-4BE40EE09EB9} | ActivateSession: Session activated
[12/15/2016 15:08:39.477] info/session  Connection 6 | SecureChannel 1 | Session {1CD777E3-9590-A4E2-6DB6-4BE40EE09EB9} | CloseSession
[12/15/2016 15:08:39.482] info/network  Shutting down the TCP network layer with 1 open connection(s)
[12/15/2016 15:08:39.498] info/network  TCP network layer listening on opc.tcp://raspberrypi:16664
[12/15/2016 15:08:39.506] info/network  Connection 6 | New connection over TCP from 127.0.0.1:40488
[12/15/2016 15:08:39.511] info/channel  Connection 6 | SecureChannel 1 | OpenSecureChannel: Opened SecureChannel
[12/15/2016 15:08:39.518] info/session  Connection 6 | SecureChannel 1 | Session {DEAFD023-EEBD-A56E-2AF2-FFDF3313F18F} | ActivateSession: Session activated
[12/15/2016 15:08:39.528] info/session  Connection 6 | SecureChannel 1 | Session {DEAFD023-EEBD-A56E-2AF2-FFDF3313F18F} | CloseSession
[12/15/2016 15:08:39.534] info/channel  Connection 6 | SecureChannel 1 | CloseSecureChannel
[12/15/2016 15:08:39.538] info/network  Shutting down the TCP network layer with 1 open connection(s)
100%: Checks: 2, Failures: 0, Errors: 0
tests/check_client_subscriptions: ====OK===

 tests/check_nodestore
[... 56 lines cut ...]
tests/check_types_builtin: ====OK===

 tests/check_types_memory
Running suite(s): testMemoryHandling
100%: Checks: 805, Failures: 0, Errors: 0
tests/check_types_memory: ====OK===

 tests/check_types_range
Running suite(s): Test Variant Range Access
100%: Checks: 4, Failures: 0, Errors: 0
tests/check_types_range: ====OK===

 tests/check_utils
Running suite(s): Utils
100%: Checks: 3, Failures: 0, Errors: 0
tests/check_utils: ====OK===
```
Also, when I run it via 'make', it is OK:
```
 make check_types_builtin
[ 77%] Built target open62541-object
[100%] Built target check_types_builtin
pi@raspberrypi:~/open62541/build $ echo $?
0
```

miraenator, 2016-12-15:
Some info for the 'valgrind issue': 

```
$ /usr/bin/ctest -VV --force-new-ctest-process check_types_builtin
UpdateCTestConfiguration  from :/home/pi/open62541/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/home/pi/open62541/build/DartConfiguration.tcl
Test project /home/pi/open62541/build
Constructing a list of tests
Done constructing a list of tests
Checking test dependency graph...
Checking test dependency graph end
test 1
      Start  1: types_builtin

1: Test command: /usr/bin/valgrind "--error-exitcode=1" "--quiet" "--trace-children=yes" "--leak-check=full" "/home/pi/open62541/build/tests/check_types_builtin"
1: Test timeout computed to be: 9.99988e+06
1: Running suite(s): Built-in Data Types 62541-6 Table 1
1: disInstr(arm): unhandled instruction: 0xF4A174DD
1:                  cond=15(0xF) 27:20=74(0x4A) 4:4=1 3:0=13(0xD)
1: ==15599== valgrind: Unrecognised instruction at address 0x4844a78.
1: ==15599==    at 0x4844A78: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem.so)
1: ==15599== Your program just tried to execute an instruction that Valgrind
1: ==15599== did not recognise.  There are two possible reasons for this.
1: ==15599== 1. Your program has a bug and erroneously jumped to a non-code
1: ==15599==    location.  If you are running Memcheck and you just saw a
1: ==15599==    warning about a bad jump, it's probably your program's fault.
1: ==15599== 2. The instruction is legitimate but Valgrind doesn't handle it,
1: ==15599==    i.e. it's Valgrind's fault.  If you think this is the case or
1: ==15599==    you are not sure, please let us know and we'll try to fix it.
1: ==15599== Either way, Valgrind will now raise a SIGILL signal which will
1: ==15599== probably kill your program.
1: ==15599== 
1: ==15599== Process terminating with default action of signal 4 (SIGILL)
1: ==15599==  Illegal opcode at address 0x4844A78
1: ==15599==    at 0x4844A78: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem.so)
 1/18 Test  #1: types_builtin .........................***Exception: Illegal  8.40 sec
```
Running valgrind with --verbose:
Note: Some mismatched CRCs. Not sure how important this is...

```
$ /usr/bin/valgrind "--error-exitcode=1" "--verbose" "--trace-children=yes" "--leak-check=full" "/home/pi/open62541/build/tests/check_types_builtin"
==15624== Memcheck, a memory error detector
==15624== Copyright (C) 2002-2011, and GNU GPL'd, by Julian Seward et al.
==15624== Using Valgrind-3.7.0 and LibVEX; rerun with -h for copyright info
==15624== Command: /home/pi/open62541/build/tests/check_types_builtin
==15624== 
--15624-- Valgrind options:
--15624--    --suppressions=/usr/lib/valgrind/debian-libc6-dbg.supp
--15624--    --error-exitcode=1
--15624--    --verbose
--15624--    --trace-children=yes
--15624--    --leak-check=full
--15624-- Contents of /proc/version:
--15624--   Linux version 4.4.36+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #933 Fri Dec 2 21:57:25 GMT 2016
--15624-- Arch and hwcaps: ARM, ARMv6-vfp
--15624-- Page sizes: currently 4096, max supported 4096
--15624-- Valgrind library directory: /usr/lib/valgrind
--15624-- Reading syms from /home/pi/open62541/build/tests/check_types_builtin (0x10000)
--15624-- Reading syms from /lib/arm-linux-gnueabihf/ld-2.19.so (0x4000000)
--15624--   Considering /lib/arm-linux-gnueabihf/ld-2.19.so ..
--15624--   .. CRC mismatch (computed 5bf4e8c8 wanted 7d09b4dd)
--15624--   Considering /usr/lib/debug/lib/arm-linux-gnueabihf/ld-2.19.so ..
--15624--   .. CRC is valid
--15624-- Reading syms from /usr/lib/valgrind/memcheck-arm-linux (0x38000000)
--15624--   Considering /usr/lib/valgrind/memcheck-arm-linux ..
--15624--   .. CRC mismatch (computed 490a0e7e wanted 8f7ca175)
--15624--   Considering /usr/lib/debug/usr/lib/valgrind/memcheck-arm-linux ..
--15624--   .. CRC is valid
--15624--    object doesn't have a dynamic symbol table
[... 89 lines cut ...]
==15624==    definitely lost: 0 bytes in 0 blocks
==15624==    indirectly lost: 0 bytes in 0 blocks
==15624==      possibly lost: 0 bytes in 0 blocks
==15624==    still reachable: 2,814 bytes in 115 blocks
==15624==         suppressed: 0 bytes in 0 blocks
==15624== Reachable blocks (those to which a pointer was found) are not shown.
==15624== To see them, rerun with: --leak-check=full --show-reachable=yes
==15624== 
==15624== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
==15624== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
Illegal instruction

$ echo $?
132
```
Info:
```
$ cat /boot/.firmware_revision 
af9cb14d5053f89857225bd18d1df59a089c171e
$ uname -a
Linux raspberrypi 4.4.38+ #938 Thu Dec 15 15:17:54 GMT 2016 armv6l GNU/Linux
$ valgrind --version
valgrind-3.7.0

```

jpfr, 2016-12-15:
It seems as if valgrind support an arm only recently fixed some errors concerning opcodes of memory operations.

https://sourceforge.net/p/valgrind/mailman/message/34752898/
http://stackoverflow.com/questions/20066215/valgrind-unrecognizes-memcmp-instruction-in-raspberry-pi

This seems to be valgrind and raspi related. Please reopen if the fixes mentioned in the above links don't work out.

Anyway, thanks for reporting this.

closed 2016-12-15 by jpfr