open62541/open62541 #913: ubuntu 12.04LTS cant complie

xty19, 2016-12-13:
OS ubuntu 12.04 LTS
OS type 64-bit
gcc (Ubuntu/Linaro 4.6.3-1ubuntu5) 4.6.3
cmake version 3.3.2
i use command to compile `gcc -std=99 open62541.c read_xml_xx2.c -o read_xml_xx2`

log `/tmp/ccAx2fYL.o:In function "UA_DateTime_nowMonotonic":`
log `(.text+0x39e5):undefined reference to "clock_gettime"`
log `collect2: ld returned 1 exit status``

and i cant compile simple server and client code.

nikita158, 2016-12-13:
@xty19 
try add `-lrt` to gcc arguments

xty19, 2016-12-14:
@nikita158 
thanks ,but it still has the problem 
:sob:

jpfr, 2016-12-15:
Did you put `-lrt` after the .c files that need it?
http://stackoverflow.com/questions/2418157/ubuntu-linux-c-error-undefined-reference-to-clock-gettime-and-clock-settim

xty19, 2016-12-15:
@jpfr  thanks it work  i put -lrt behind .c files  and it work!

closed 2016-12-15 by xty19