open62541/open62541 #914: Can't add variable or variableType with dataType attribute

j-sc, 2016-12-14:
Adding a variable or variableType node doesn't work if the dataType attribute is set. Works fine without the dataType.

Minimal example server:
```C
#include <signal.h>

#ifdef UA_NO_AMALGAMATION
#include "ua_types.h"
#include "ua_server.h"
#include "ua_config_standard.h"
#include "ua_network_tcp.h"
#include "ua_log_stdout.h"
#else
#include "open62541.h"
#endif

UA_Boolean running = true;
UA_Logger logger = UA_Log_Stdout;

static void stopHandler(int sign) {
	UA_LOG_INFO(logger, UA_LOGCATEGORY_SERVER, "received ctrl-c");
	running = false;
}

int main(int argc, char** argv) {
	signal(SIGINT, stopHandler); /* catches ctrl-c */

	UA_ServerConfig config = UA_ServerConfig_standard;
	UA_ServerNetworkLayer nl;
	nl = UA_ServerNetworkLayerTCP(UA_ConnectionConfig_standard, 16664);
	config.networkLayers = &nl;
	config.networkLayersSize = 1;
	UA_Server *server = UA_Server_new(config);

	// variableType
	UA_VariableTypeAttributes vtAttr;
	UA_VariableTypeAttributes_init(&vtAttr);
	vtAttr.displayName = UA_LOCALIZEDTEXT("", "TestVariableType1");
	vtAttr.description = UA_LOCALIZEDTEXT("", "");
	vtAttr.valueRank = -2;

	UA_StatusCode sc = UA_Server_addVariableTypeNode(server,
			UA_NODEID_NUMERIC(1, 1000),
			UA_NODEID_NUMERIC(0, 63),
			UA_NODEID_NUMERIC(0, 45),
			UA_QUALIFIEDNAME(1, "TestVariableType1"),
			UA_NODEID_NULL, //why do we need a typedefinition for a type?
			vtAttr, NULL, NULL);

	UA_LOG_ERROR(logger, UA_LOGCATEGORY_SERVER, "VariableType: 0x%x", sc);

	UA_VariableTypeAttributes vtAttr2;
	UA_VariableTypeAttributes_init(&vtAttr2);
	vtAttr2.displayName = UA_LOCALIZEDTEXT("", "TestVariableType2");
	vtAttr2.description = UA_LOCALIZEDTEXT("", "");
	vtAttr2.dataType = UA_NODEID_NUMERIC(0, 12);
	vtAttr2.valueRank = -2;

	sc = UA_Server_addVariableTypeNode(server,
			UA_NODEID_NUMERIC(1, 1001),
			UA_NODEID_NUMERIC(0, 63),
			UA_NODEID_NUMERIC(0, 45),
			UA_QUALIFIEDNAME(1, "TestVariableType2"),
			UA_NODEID_NULL,
			vtAttr2, NULL, NULL);

	UA_LOG_ERROR(logger, UA_LOGCATEGORY_SERVER, "VariableType with dataType: 0x%x", sc);
		

	// variable
	UA_VariableAttributes attr;
	UA_VariableAttributes_init(&attr);
	attr.description = UA_LOCALIZEDTEXT("","");
	attr.displayName = UA_LOCALIZEDTEXT("","TestVariable1");

	sc = UA_Server_addVariableNode(server,
			UA_NODEID_NUMERIC(1, 1002),
			UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
			UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
			UA_QUALIFIEDNAME(1, "TestVariable1"),
			UA_NODEID_NUMERIC(0, 63),
			attr, NULL, NULL);

	UA_LOG_ERROR(logger, UA_LOGCATEGORY_SERVER, "Variable: 0x%x", sc);

	UA_VariableAttributes attr2;
	UA_VariableAttributes_init(&attr2);
	attr2.description = UA_LOCALIZEDTEXT("","");
	attr2.displayName = UA_LOCALIZEDTEXT("","TestVariable2");
	attr2.dataType = UA_NODEID_NUMERIC(0, 12);

	sc = UA_Server_addVariableNode(server,
			UA_NODEID_NUMERIC(1, 1002),
			UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
			UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
			UA_QUALIFIEDNAME(1, "TestVariable2"),
			UA_NODEID_NUMERIC(0, 63),
			attr2, NULL, NULL);

	UA_LOG_ERROR(logger, UA_LOGCATEGORY_SERVER, "Variable with dataType: 0x%x", sc);

	UA_StatusCode retval = UA_Server_run(server, &running);
	UA_Server_delete(server);
	nl.deleteMembers(&nl);

	return (int)retval;
}
```

jpfr, 2016-12-14:
In the second example, you define a string datatype (see here: https://github.com/open62541/open62541/blob/0.2/tools/schema/NodeIds.csv).

 It is not allowed to have an empty value if the datatype is not BaseDataType.
So you also need to fill the value attribute. There should be a somewhat descriptive error message.
Otherwise we add one.

j-sc, 2016-12-14:
Why aren't empty values of other datatypes allowed? There are a lot of variables with an empty value and different datatypes defined in NS0.

jpfr, 2016-12-14:
The standard says so.

> 8.7 BaseDataType
This abstract DataType defines a value that can have any valid DataType.
It defines a special value null indicating that a value is not present.

So if you force a String datatype, you need to set a string. No value is not a string.
If you set the BaseDataType as the datatype (which we do by default if you leave the null nodeid in the attributes struct), then you can set any value (String, Boolean, ...) and also an empty value.

Can you give an example for a Variable with a datatype and no value in ns0?
We could file an issue in the OPC Foundation bugtracker and get more clarification.

jpfr, 2016-12-14:
Hmm. The standard defines a second mechanism for (dis-)allowing null values.
That gives us a bit more leeway for our implementation.

> The AllowNulls Property is only used for DataVariables. It does not
apply to Properties.
This Property specifies if a null value is allowed for the Value Attribute
of the DataVariable. If it is set to true, the Server may return null
values and accept writing of null values. If it is set to false, the Server
shall never return a null value and shall reject any request writing a
null value.
If this Property is not provided, it is Server-specific if null values are
allowed or not.

We could allow null values for every data type.
Any strong opinions within the core dev group?

j-sc, 2016-12-15:
Example from ns0: i=2761
I use variables with null values mainly as properties of types. The AllowNulls Property wouldn't help  there. Should they get the corresponding null or empty value for their datatypes assigned?
(String -> ""   |   NodeId -> ns=0,i=0   |   ...)

jpfr, 2017-01-06:
The type checking was relaxed when no value is present.
So now you just get a "BadNodeExists" on the second try.

Best regards,
Julius

commit 22f42198 (2017-01-06): relax type checking when no value is present (fix #914)
commit 8285627b (2017-01-08): relax type checking when no value is present (fix #914)
closed 2017-01-06 by jpfr