open62541/open62541 #915: Determine if OPC_UA node is folder, tag, or function

tfmeads, 2016-12-14:
I'm writing a back-end library for a OPC UA browser which implements open62541, and I need a function that will let me know if the given node is a folder, a function, or a tag with data. Currently I'm just reading the value of the node then checking the value with UA_DataValue.hasValue(), if it's false it must be a folder or a function but I don't know which one it is. Is there a function contained within the open62541 library that will allow me to check this directly, and if not, what's the best approach for doing so?

Thank you in advance

jpfr, 2016-12-14:
You can read the `NodeClass` attribute.
That tells you which of the 8 possible node classes you have.

Note that there is no "folder" node class. Only `ObjectNode`.
Then, you can browse the references from that node to find whether it has a `hasTypeDefinition` reference to the `FolderType`.

Or checking for objects is already enough for you...

tfmeads, 2016-12-15:
Ok, that makes sense. If I check the node and see if it's nodeClass == UA_NODECLASS_OBJECTTYPE, it's an ObjectNode, and has the potential to be a folder, correct? The only thing I'm confused about is "hasTypeDefinition". I can't find a reference to a method or variable called that anywhere in the documentation. How do I browse the references from a given node?

jpfr, 2016-12-15:
Currently, the best spec for the browse service is the standard itself (part 4).
Here's the API for the client: http://open62541.org/doc/current/client.html#view-service-set

In short, nodes in OPC UA are connected via typed references that can be seen as triples (source, predicate, target). The browse service returns a (filtered) list of the references starting from some node.
The `hasTypeDefinition` is a nodeid of a ReferenceTypeNode and used for the reference predicate (or reference type). The `FolderType` is the nodeid of an ObjectTypeNode.

I hope this gives you some pointers for looking at the standard + our documentation.

tfmeads, 2016-12-15:
That clears it up - I think that answers my question. Thank you very much!

closed 2016-12-15 by tfmeads