open62541/open62541 #916: consequences of subscription handling on datasource variable node's?

caspervanzoest, 2016-12-15:
I currently have an open62541 server running with datasource variable node's.
Whenever a client reads a (node)value, the (node)value will be updated with the current (source)value of the source.

_Reason for this setup is,
I am running an internal 'database' which is constantly updated.
I am making this database more accessible trough OPCUA._

Any client can subscribe to the database (source)values in order to be notified when a (node)value has been changed.
(either from another client, or from the internal 'database' where the server is linked to.

Because the server is linked to another internal 'database',
The only way to update the (node)values on the server, is by being read from a client.
Reason for this is not having to update all node's on the server, even without being used by any client.

This leaves me with the questions:
1. How are client subscriptions to datasource implemented?
2. Are they 'polling' the datasource in order to receive a data-change event?
3. Is the 'check' for data-changed inside the server or inside the client?

_extra:
4. Are both 'reporting' and 'sampling' mode available on a certified client to the open62541 server?
5. Are both 'reporting' and 'sampling' mode available on a open62541 client to the open62541 server?_

Any suggestion is welcome. Thanks for answering!

jpfr, 2016-12-18:
1. The *DataChange* MonitoredItems use the read service. So the call to the datasource is identical to a read request.
https://github.com/open62541/open62541/blob/0.2/src/server/ua_subscription.c#L147

2. Yes. It's like polling the read service. But done internally.

3. Inside the server

4. and 5. Reporting and sampling are related to the triggering mechanisms where some samples are only reported when an external condition (in another monitoreditem is triggered.
The open62541 server does currently not support triggering. The open62541 client can use triggering in other servers.

caspervanzoest, 2016-12-19:
Thanks alot!

closed 2016-12-19 by caspervanzoest