open62541/open62541 #918: Faster way to discern variable's data type?

tfmeads, 2016-12-22:
Hello,

I've written a simple OPCUA file browser using open62541. Performance is great except when I browse a node with a large amount of children- when I browse a node with 1000+ children it regularly takes over 10 seconds to process them all. After looking into it, I found that the bottleneck was in the part of my process where I perform a UA_Client_service_read() on every node to discern its data type value, if it has one (without this part of the process, it takes less than 2 seconds to process 1000+ nodes).

Here's my code:

    UA_ReadResponse rResp = UA_Client_Service_read(*/reference to desired node/*);
    
    if (rResp.responseHeader.serviceResult == UA_STATUSCODE_GOOD && rResp.resultsSize > 0 &&
        rResp.results[0].hasValue && UA_Variant_isScalar(&rResp.results[0].value)){ //if read is good and result is not null and not an array

    //do things based on value of rResp.results[0].value.type

Is there any other way to discern the type of a given variable node's data? It's clear that using the read service is slowing things down quite considerably.

jpfr, 2016-12-23:
What is taking time is probably not the reads per se but the 1000+ network roundtrips.

Have a look at `UA_Client_Service_read`. That's the reaw read service.
And it contains an array of several ReadValueId members. So you can squeeze many reads in a single TCP packet.

tfmeads, 2016-12-23:
That solved the problem wonderfully. Thank you!

closed 2016-12-23 by tfmeads