open62541/open62541 #921: Server_NamespaceArray type seems wrong

oroulet, 2016-12-27:
Description
===========

Server_NamespaceArray node has data type Null. Shouldn't be String?

Then trying to write to it from client I get an an Error: 

BadTypeMismatch: The value supplied for the attribute is not of the same type as the attribute"s value.(BadTypeMismatch)

I want to write to this variable to add open62541 supoprt in opcua-modeler

Background Information / Reproduction Steps
===========================================

See here for the kinds of information we need to reproduce the bug and how to obtain it:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Checklist
=========
Please provide the following information:

 - [ master ] open62541 Version (release number or git tag):
 - [several ] Other OPC UA SDKs used (client or server): 
 - [linuc ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

closed 2017-01-01 by jpfr

oroulet, 2017-01-01:
> The namespacearray datatype was added to 0.2 and master.

Does that fix writing to the node from client? From a client I could change
data type but, even after change, I could not write to it. Reproductible
with uaexpert

jpfr, 2017-01-01:
It does not fixing writing into the node.
The namespacearray node has a datasource where writing is not implemented.
https://github.com/open62541/open62541/blob/0.2/src/server/ua_server.c#L926

The server API let's you add (but not remove) namespaces manually.
https://github.com/open62541/open62541/blob/0.2/include/ua_server.h#L142

If you have a use case where an external client has to edit the namespacearray, we can add writing to the datasource.

oroulet, 2017-01-01:
The use case is opcua-modeler, but the behov will come up within any GUI. User expect the possibility to add/remove edit values.
In my case I could survive with a delete method ..

Stasik0, 2017-01-01:
So users need to add namespaces dynamically at runtime

oroulet, 2017-01-01:
I guess so. At least I need it in opcua-modeler since I use an internal
server to ensure consistent address space, and hopefully later scripting
server side or from client.

jpfr, 2017-01-06:
You can now add namespaces by writing a namespacearray.
But deleting namespaces is still not supported.
That could lead to a whole lot of complications once we attach different NodeStores to namespaces (currently in a feature branch).

If that comes up *a lot* in your tool, give us a hint.

oroulet, 2017-01-06:
If I can write an array then I can delete a namespace, can't I? At least
rename ?

Btw where is the xml import code/function? I read somewhere there was one
but I could not find it...I can anyway use my python client code but if you
have an import function I should use it.

oroulet, 2017-01-06:
Btw rename is enough I guess

commit 9ebd645c (2017-01-01): set namespacearray type (fix #921)