open62541/open62541 #925: Modifying namespace array for GUI

oroulet, 2017-01-07:
To continue discussion from https://github.com/open62541/open62541/issues/921 
To let user edit namespaces in a GUI in a user friendly way we ideally need to let user add, delete and modify namespaces.
@jpfr  thanks to your commit https://github.com/open62541/open62541/commit/2a93aebd3925fa8cd0df1a0c92e78d418511858e I can now add a namespace. But I would like to be also able to edit a namespace. Could this be enabled? With the current solution it is still possible to fuck up everything anyway since I can reorder namespaces.

jpfr, 2017-01-07:
Okay. I then add a flag "prototypeMode" or similar to the server configuation.
The flag gives additional write permissions (such as writing namespaces) that we don't want to expose in production.

The flag will be added to the master branch, as the server configuration struct is part of the public API that is frozen for 0.2.

oroulet, 2017-01-07:
Sounds good!

JGrothoff, 2017-02-06:
Hey, in my PR #832 . I also added a "UA_Namespace" with the corresponding functionality. There you ca easily add/delete/reorder namespaces from a client.

Therefore all nodes and references pointing to a updated namespace are updated.

closed 2017-02-08 by jpfr