open62541/open62541 #928: Duplicate and missing objects on different namespaces from nodeset generated models

XB32Z, 2017-01-09:
Hi all, 

I am working on a server on rapsberry pi to connect an IMU. I am modeling the measured gyroscope values in a CubeItemType but it doesn't appear in the client (UAExpert). Also the rest of the attributes are shown two times: once in namespace 1 and once in namespace 2. 

I created a IMUType (in UAModeler) that has the properties: Communication, Manufacturer and the variable MeasuredGyro of type CubeItemType. I instanciate it in the objects and from the code I feed the value. (Of course I am using a parsed xml into nodeset etc.) 

Did I missed something while setting up the nodeset?

Thx for the help,

Checklist
=========
 - v0.2
 - Client is UAExpert
 - Debian for raspberry Pi

FlorianPalm, 2017-01-09:
Hi,
the first issue (duplicated objects) is known. If you create an instance within your nodeset c-code is generated (Python-script) to create this object and its child-nodes (namespace 2). Furthermore, the open62541-stack generates instantiation code which also creates child-nodes of this object (namespace 1).
A workaround would be to create instances at runtime and not within the nodeset.
Could you show some of your xml-code for your CubeItemType. I guess you have used a VariableType to express your data rather than creating a new Datatype, correct?
Best regards,
Florian

XB32Z, 2017-01-09:
Hi, 
Thx for the quick answer :D 
Ok for the duplicated item!
Yes I am using a VariableType, although I could use a DataType. I though that it made more sense. You can find attached the xml of it. 
Thx again,
Pierre

[mpu9250.txt](https://github.com/open62541/open62541/files/693974/mpu9250.txt)

jpfr, 2017-01-09:
Hey Pierre,

that's a mismatch of the nodeset compiler and the instantiation code.

During instantiation, we create all children of a variabletype automatically.
Then the nodeset compiler additionally creates the manually defined child nodes.

This is solved in PR #919. I will give you a quick heads-up when the branch has landed.

XB32Z, 2017-01-10:
Hi, 
Do you mean, the doubled objects? Instantiating them in the code is not a problem, but the missing objects (CubeItemType) is more problematic. I compiled it with trace level of log and I got the BadReferenceTypeIdInvalid error. 
Thx for your support, 
Pierre

closed 2017-01-12 by XB32Z