open62541/open62541 #931: client state is not updated in function UA_Client_disconnect

murrayzhong, 2017-01-16:
Description
===========
There is a bug in the function UA_Client_disconnect().  After invocation of the function, the state of client is still CONNECTED. As a result if you want to execute code like:
    while(1){
        UA_Client_conenct(client, ENDPOINT_URL);
        UA_Client_readValueAttribute(...);
        UA_Client_disconnect(client);
   }
Function UA_Client_readValueAttribute() will return state code "0x80ae0000" which is "The network connection has been closed". The reason is that in the disconnect function, client state has not been updated after the connection is disconnected. We can solve this issue by just simply add code to update the state: 

UA_StatusCode UA_Client_disconnect(UA_Client *client) {
    ...

   //add code at the end of this function 
    if(client->connection->state == UA_CONNECTION_CLOSED)
    {
    	client->state = UA_CLIENTSTATE_READY;
    }
    return retval;
}

Background Information / Reproduction Steps
===========================================

See here for the kinds of information we need to reproduce the bug and how to obtain it:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): abc3716
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Root-Core, 2017-01-30:
Could have to do with PR #855. There is definitely something faulty.

murrayzhong, 2017-02-01:
/*
 *  Use the following code to reproduce this issue.
 */
#include <stdio.h>
#include "open62541.h"
#include <sys/time.h>

#define ENDPOINT_URL                    "opc.tcp://192.168.0.35:16664"  // <=== my opc-ua server

void ReadData(UA_Client *client, UA_NodeId nodeId)
{
    UA_StatusCode status;
    UA_NodeId dataType;
    UA_NodeId_init(&dataType);

    status = UA_Client_readDataTypeAttribute(client, nodeId, &dataType);
    if(UA_STATUSCODE_GOOD != status)
    {
       	printf("Err: UA_Client_readDataTypeAttribute()->status=%x =>", status);
       	if(UA_STATUSCODE_BADCONNECTIONCLOSED == status) //0x80ae0000
       		puts("The network connection has been closed");
    }
}

int main(int argc, char *argv[])
{
    const UA_NodeId nodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERSTATUS);
    UA_StatusCode status;
	UA_Variant value;
    UA_Client *client = UA_Client_new(UA_ClientConfig_standard);
    if(NULL == client) return -1;

    while(1)
    {
        status = UA_Client_connect(client, ENDPOINT_URL);
        printf("\nConnect to %s...%s\n", ENDPOINT_URL, UA_STATUSCODE_GOOD == status ? "OK" : "Failed");
    	if(status != UA_STATUSCODE_GOOD) goto LABEL_QUIT;

    	ReadData(client, nodeId);

       	if((status = UA_Client_disconnect(client)) != UA_STATUSCODE_GOOD)
       	{
            printf("Error: Client_disconnect() failure (%x)", status);
            if(UA_STATUSCODE_BADNOTCONNECTED == status)
            {
            	printf("UA_STATUSCODE_BADNOTCONNECTED");
            }
            puts("");
       	}
        sleep(1);
    }

LABEL_QUIT:
    UA_Variant_deleteMembers(&value);
  	if(client) UA_Client_delete(client);

  	return 0;
}

My display:
Connect to opc.tcp://192.168.0.35:16664...OK

Connect to opc.tcp://192.168.0.35:16664...OK
Err: UA_Client_readDataTypeAttribute()->status=80ae0000 =>The network connection has been closed
Error: Client_disconnect() failure (808a0000)UA_STATUSCODE_BADNOTCONNECTED

Connect to opc.tcp://192.168.0.35:16664...OK

Connect to opc.tcp://192.168.0.35:16664...OK
Err: UA_Client_readDataTypeAttribute()->status=80ae0000 =>The network connection has been closed
Error: Client_disconnect() failure (808a0000)UA_STATUSCODE_BADNOTCONNECTED
......

jpfr, 2017-02-05:
PR #855 fixed the issue with the example code.

closed 2017-02-05 by jpfr