open62541/open62541 #933: Cannot authenticate to Siemens 840D

xcvb85, 2017-01-17:
I cannot authenticate to Siemens 840D control. Anonymous connection works well but I need to to login to get read access to the real data (more then just date/time).

Version: master from today (git tag e1cfdc0)
OS: Debian Stretch
Server: Siemens 840D SL V4.7

This is my little test program:
```
#include "open62541.h"

int main(int argc, char *argv[])
{
    UA_Client *client = UA_Client_new(UA_ClientConfig_standard);

    UA_StatusCode retval = UA_Client_connect_username(client, "opc.tcp://192.168.214.1:4840", "OpcUaClient", "SUNRISE");

    if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(client);
        return (int)retval;
    }

    UA_Client_disconnect(client);
    UA_Client_delete(client);
    return (int) UA_STATUSCODE_GOOD;
}
```
Console output:
> [01/17/2017 15:10:29.866] debug/network	Sent HEL message
> [01/17/2017 15:10:29.866] debug/network	Received ACK message
> [01/17/2017 15:10:29.866] debug/channel	Requesting to open a SecureChannel
> [01/17/2017 15:10:29.866] debug/channel	SecureChannel opened
> [01/17/2017 15:10:29.866] debug/client	Sending a request of type 426
> [01/17/2017 15:10:29.866] debug/client	Received a response of type 431
> [01/17/2017 15:10:29.866] error/client	No anonymous token found

jpfr, 2017-01-17:
Hello @xcvb85,

it is possible that username/password login is only possible with encryption enabled.
open62541 currently does not support encryption.
Can you have a look with UAExpert which Endpoints the server has?

Some servers require the password to be encrypted for login, even if the securechannel itself is not encrypted. This does not show in UAExpert however.

xcvb85, 2017-01-17:
Hello @jpfr 

sorry I'm too stupid to find the endpoints in UaExpert so I used wireshark:
> endpoint 0: SecurityPolicy#none
> endpoint 1: SecurityPolicy#Basic128Rsa15
> endpoint 2: SecurityPolicy#Basic256

The connection with other OpcUa-clients works without encryption but I get the following error messages:

UAExpert:
> Register for ModelChangeEvents failed: createMonitoredItems returned BadFilterNotAllowed for item 0

Softing Client:
> ERROR - Endpoint Server ApplicationUri is not the same with ServerCertificate ApplicationUri. Mask:ClientAPI

Maybe the Siemens implementation isn't perfect either.

jpfr, 2017-01-17:
Every endpoint contains a list of UserTokenPolicys.
The TokenPolicys may require encryption of the password even though the SecureChannel is unencrypted.

Since you made Wireshark logs anyway, can you look at the SecurityPolicy#none endpoint in detail?

jpfr, 2017-01-18:
Thanks for confirming.
I added a feature-issue #934 to track the eventual development of the feature.

closed 2017-01-18 by jpfr