open62541/open62541 #934: Support UserTokenPolicys with encryption

jpfr, 2017-01-18:
Endpoints for Securechannels with SecurityPolicy#None can still mandate that passwords are transmitted encrypted. This is specified in the list of possible UserTokenPolicys for each endpoint.
This issue is used to track the development of the feature.

Issues where clients cannot connect because of encryption UserTokenPolicys are closed as duplicates.

Pro, 2018-02-01:
Related: #1539

patrickjuchli, 2018-11-06:
In the referenced issues I found the suggestion to "switch to encrypted endpoint" for a fix. This is a bit misleading. If the server requires user tokens to be encrypted, connecting to it fails regardless of whether encryption is used for the endpoint or not. This makes sense when you read the following lines: https://github.com/open62541/open62541/blob/43a2b539c2befd970edb121a5df706f130efd22c/src/client/ua_client_connect.c#L456-L458

I can see that this feature is part of the next milestone, thanks for that!

sureshaei, 2018-12-26:
We are using open62541 (master) client to connect to a PLC and we get the error "No suitable UserTokenPolicy found for the possible endpoints" when using username and password.

We tried using the example clients with encryption enabled (client_basic128rsa15). We created client certificate.crt and private-key.key files using openssl. We converted the files to .der files and provided as arguments to the client. It returns error "mbedTLS returned an error: PK - Invalid key tag or value; Could not create securityContext".

jpfr, 2018-12-27:
Can you try to generate cert/key with the /tools/cert/create_certificate.py script provided with open62541?

sureshaei, 2019-01-03:
Thanks for the reply.
We tried creating .der files as described above. The client application is able to connect with the server.

We tried replacing the UA_client_connect() with UA_client_connect_username() in the same example. We got end up with " error/client	No suitable UserTokenPolicy found for the possible endpoints"

May i know whether the support for accessing OPC UA server using username password (encrypted) added to the stack as of now?

heppth, 2019-01-04:
Hi, I'm monitoring this issue for about 8 months. Is there the change that user / password authentication with encryption will be implemented soon? Most of the PLCs can not be connected with this library. For us, the function is very important.
Thank you

jpfr, 2019-02-10:
Implemented on master.

closed 2019-02-10 by jpfr

reyanvaldes, 2019-05-17:
Hi, thank you so much for sharing this amazing complex work you guys are doing, and supporting. 
I am facing since several days same problem trying to connect to Kepware V6 when using username & password with security police #None. If I am not using username/password, it is ok. I tried to include the supported security policies as recommended, but no success. I got message 'Could not create securityContext', I am using the ca.crt and ca.key generated by the self-signed tool provided. Attached is section of code in C++ and screenshot. Please could you recommend where to looks, complete or what am I doing wrong? Thank you so much.
[image]

[connect_username_password_kepware.txt](https://github.com/open62541/open62541/files/3192963/connect_username_password_kepware.txt)