open62541/open62541 #938: Open62541 server cannot register to discovery server from OPC Foundation

XB32Z, 2017-01-19:
Hi all, 

I am trying to register a server to the OPC Foundation LDS but it doesn't work. The first problem was that it didn't accept unsecure connection but I changed this into the config file of the discovery server. Now I have the problem that the service required by my server is not recognise. 
I receive from the LDS the error code 0x800b000 UA_STATUSCODE_BADSERVICEUNSUPORTED which I can see in the output of the LDS as "received request for unsupported service"

Checklist
=========
Please provide the following information:

 - [master ] open62541 Version (release number or git tag):
 - [LDS from OPC Foundation ] Other OPC UA SDKs used (client or server): 
 - [Windows & Unix ] Operating system:

jpfr, 2017-01-23:
@Pro can you check?

Pro, 2017-01-23:
@XB32Z you say that you used the master branch?
LDS is not yet in the master branch, you need to use the feature_mdns for full LDS support:
https://github.com/open62541/open62541/tree/feature_mdns

Can you try with that one?

XB32Z, 2017-01-24:
Hi @Pro , 
I just tried with the feature mdns but it doesn't work either. When I use the normal parameter of the LDS, as it hasn't the security#None policy the open62541 cannot register because of not found endpoint. When I add the security#None policy, the LDS complains that the requested service is not supported

Thx

Pro, 2017-01-24:
Ok, I'll try to investigate that further. But don't expect a response too soon since I'm currently highly involved in another project at work.
If you didn't find them yet, there are also specific examples for the discovery service in open62541:
https://github.com/open62541/open62541/tree/feature_mdns/examples/discovery

Can you post your OPC Foundaction LDS config file which you use?

jpfr, 2017-02-20:
Hey @Pro ,

do we have a status on this?

Pro, 2017-02-20:
I'll have a look at it.

@XB32Z can you post your OPC Foundation LDS config file?

Pro, 2017-02-20:
Fixed in https://github.com/open62541/open62541/commit/9b00e647c892f96cec91905dd64892ff95e151a9

Tested with OPC foundation Server 1.03.340

The problem was that the open62541 Server/Client tried to open a session which is not allowed for FindServerOnNetwork.
Registering should have worked anyways. The only thing I saw was that version 1.03.340 does not yet support RegisterServer2, thus you will see a `BADSERVICEUNSUPPORTED` message in the output.
You can ignore this, because open62541 first tries RegisterServer2, if that fails or is unsupported, RegisterServer is tried. This works with OPC Foundation's LDS Server.

Pro, 2017-03-07:
This issue still persists.
It seems that they changed the implementation in the current version (https://github.com/OPCFoundation/UA-LDS) so that RegisterServer and RegisterServer2 require encryption (see https://github.com/OPCFoundation/UA-LDS/issues/1).

So to finally fix this, we need encryption in open62541

XB32Z, 2017-03-08:
Hi, 
Do you need any input from me?

Pro, 2017-03-08:
No, not for now. We have to wait until open62541 supports encryption to be able to connect to the OPC Foundation's LDS Server.

It is currently planned but we don't have any release date yet.

syvaren, 2017-08-03:
I have tasks to authenticate and encrypt the opcua server I am working on.  I was wondering about the status of encryption support for open62541.

Looking at master branch the ServerConfig has a UA_ByteString serverCertificate.  Is this used to authenticate connections automatically or do I need to create an activateSession function to intercept and perform authentication on the server side?

On the client side I do not see how to send the certificate for authentication, how would I go about doing this?

Pro, 2017-08-04:
@syvaren unfortunately encryption is not yet in the master branch, but is near it's finishing point and will be integrated soon (see #1003). I did not test it in combination with the OPC Foundation LDS, but theoretically it should work.

We are always happy if community members are willing to help! You could ask in #1003 if any help is needed, e.g. for testing. Another good starting point for helping us and to get familiar with the source code is to look at the issues list and pick issues with `developer-wanted` and `easy-hack`:
https://github.com/open62541/open62541/issues?q=is%3Aopen+is%3Aissue+label%3Adeveloper-wanted

An example for the server certificate can be found here:
https://github.com/open62541/open62541/blob/master/examples/server_certificate.c

This certificate will be sent for the CreateSession response and is added to the endpoint definitions.
This means that a client can check the certificate and thus validate if it is connecting to the correct server (similar to HTTPS).

The certificate is not used for authentication. This will come as a follow-up of the encryption plugin.

Infinity95, 2017-08-04:
Do keep in mind though, that #1003 only implements the plugin architecture that is necessary for encryption, and only on the server side. A plugin for the SecurityPolicy#Basic128Rsa15 is available on my fork, although it's unfinished.

jpfr, 2018-01-17:
Encryption is ready now.
What is required to close this?

Pro, 2018-01-17:
Nice!! In summary I need to test if everything works as it should with the OPC Foundation LDS.
I will check that in the coming days and finally close the issue.

Pro, 2018-01-19:
@Infinity95 do we also already have encryption (`SignAndEncrypt`) for the client side?

The OPC Foundation LDS requires `Basic128Rsa15` with probably Sign or SignAndEncrypt to call RegisterServer

Infinity95, 2018-01-19:
No, the client side doesn't support encryption yet. It uses some parts of the new plugin interface already, but still needs changes to actually support it.

commit 9b00e647 (2017-02-20): Client must not create a session for FindServer calls.

Non-Multicast functionalities tested with OPC Foundation LDS Server. Fix #938
commit 5b14c838 (2018-03-26): Support OPC Foundation LDS server

Since client encryption landed into master, this was an open todo for
a long time. The multicast server example was updated to use the encryption
client.

Fixes #938
commit c98e107c (2018-03-26): Support OPC Foundation LDS server

Since client encryption landed into master, this was an open todo for
a long time. The multicast server example was updated to use the encryption
client.

Fixes #938
commit 4960fd8b (2018-03-27): Support OPC Foundation LDS server

Since client encryption landed into master, this was an open todo for
a long time. The multicast server example was updated to use the encryption
client.

Fixes #938
closed 2018-03-28 by Pro