open62541/open62541 #940: Browse Request failed if more than 2 TCP segments

robot1111, 2017-01-20:
Hello
i am currently implementing an OPC UA server on our embedded system. We use our own multithreaded OS running on an Cortex M3.
Current used tag: c942f26d5f 17.1.2017

If I try to browse a small subtree (see appended .jpg below) then it works.
If I try to browse a bigger subtree then it doesn't work. You can't see the content of the subtree in UaExpert.
The difference is that the Browse Response is fragmented in the latter case. You can see this fragmantation in the appended WireShark Trace starting with frame 159.
I also appended a log file output via telnet (UA_LOGLEVEL 100).

Best
Martin

[image]
[3 segments fail frame 160.zip](https://github.com/open62541/open62541/files/719664/3.segments.fail.frame.160.zip)
[3 segments fail.zip](https://github.com/open62541/open62541/files/719666/3.segments.fail.zip)

jpfr, 2017-01-21:
Hello Martin,

looking at the Wireshark dump there are some observations to be made:

- There are three messages of length 1514 (with payload length 1460) and none above that. It could be that the TCP implementation on the Cortex splits packets at that length due to intenal buffer limitations  [1].
- Around packet 29, this splitting and reassembling works fine. But there is only one "big" PDU.
- The payload of packets 159/160 is a message of length 3132 (0x00000c3c). But only 2x1460 bytes are transmitted. Then the clients sends an ACK and the server terminates the connection. Probably the termination happens in the network layer and is not explicitly triggered in "our" part of the server.
- OPC UA can chunk messages (split the payload into several messages) internally. But that has not happened here, as the packet length needs to be larger than 8192 bytes for that.

Since no chunking happens, the server did put all the 3132 bytes of the payload in a single buffer to be sent via TCP. That succeeded, because no error was logged. But less than 3132 bytes arrived at the client. I would either check the TCP implementation, the possiility of packet loss **or** just use the browse service with less elements. I don't know about UAExpert, but the browse service has a `requestedMaxReferencesPerNode` argument. If there are more references you get back the identifier of a *ContinuationPoint*. The BrowseNext request can be used to get the next batch of references for the ContinuationPoint.

[1] TCP is a streaming protocol and may merge and split packets as it wishes to. As long as the order remains correct.

jpfr, 2017-01-28:
Hello Martin,
did you dig any deeper into this?
Can we close?

robot1111, 2017-01-30:
Hello Julius,
meanwhile I could fix this issue. The problem was caused by an incorrect evaluation of the error state of our TCP/IP stack. The "Would Block" notification of the TCP/IP stack was missed resp. not correctly handled in socket_write. Our system ran without any problems over this weekend. Thus I think that we can close this issue.
Thanks for your fast response und support.
Martin

jpfr, 2017-01-30:
Great. I'm glad it worked out for you.

closed 2017-01-30 by jpfr